PX4/PX4-Autopilot #19288: No Airspeed Sensor Found on Master Build

ttropcic, 2022-03-07:
Since I had occasional problems with Airspeed Sensor Failure(-1,1) in flight on the latest stable build, I've tried switching to latest master (1.13.0 alpha), as I've seen someone else report the same issue on landing. According to https://github.com/PX4/PX4-Autopilot/pull/18442 , it should've somehow solved the issue.

When using the stable release, I have no problem with detecting and calibrating the airspeed sensor.
But after switching to Master, I get the "No Airspeed Sensor found".

Hardware: - Hex / Profi CNC Cube Orange flight controller
                 - PX4 Airspeed v1.3, like in the link:  https://www.mybotshop.de/Pixhawk-Digital-Airspeed-Sensor

Maybe it's some of Cube Orange's driver issue? Or am I just missing something in the setup?

Thanks.

dagar, 2022-03-07:
For various reasons these differential pressure (airspeed) sensors now need to be enabled explicitly.

Can you try setting `SENS_EN_MS4525 1`?

ttropcic, 2022-03-07:
Awesome, fixed.

Thanks!

nodes82, 2022-03-10:
It's been a big help.
How do I use the GIT command to download the PX4 stable version?
thank you.