PX4/PX4-Autopilot #19289: DSM not works with v1.12.3, but works with 1.11.3 PixRacer

lobart, 2022-03-07:
I updated my FC Pixracer to v1.12.3, and when i go to Radio in QGC, there are not any sound and when i click Calibrate, it send me - Detected 0 radio channels. To operate PX4, you need at least 5 channels.
But with 1.11.3 all is ok.

Please, help!

dagar, 2022-03-07:
What hardware do you have exactly? Specific Spektrum receiver and transmitter.

lobart, 2022-03-07:
I have custom DSM, made by self.
But it works with 1.11.3 and previous versions of PX4.
With versions 1.12.0 and older it is not working.
What is important logic difference between this versions? 
With 1.12.0 not messages by "listener input_rc".

dagar, 2022-03-07:
Getting solid DSM parsing across the full range of legitimate Spektrum and 3rd party DSM hardware has been a bit of a struggle.

https://github.com/PX4/PX4-Autopilot/pull/18270

Do you know what your data looks like (DSM2, DSMX, etc)? Are you populating the initial system byte?

lobart, 2022-03-07:
dsmx
shift = 11

lobart, 2022-03-07:
````c
uint16_t channelValueToBytes(uint16_t channel, uint16_t value) {
    const uint16_t shift = 11;
    uint16_t bytesPair = value;
    bytesPair |= (channel << shift);
    return bytesPair;
}

// maps initial value in [0..100] interval to a value in [0..2047)
uint16_t scaleChannel(uint16_t initial) {
    const double unscaled_max = 100., scaled_max = 2047.;
    const double factor = scaled_max / unscaled_max;
    return (uint16_t)(factor * (double)initial);
}

// NOTE: restore BIT_GO_DOWN and initial values
void dsm_control_task() {
    REGISTER_TASK(xTaskGetCurrentTaskHandle(), DSM)

    dsm_packet[0] = 0x00;
    dsm_packet[1] = 0xA2;
    dsm_packet[2] = 0x0B;
    dsm_packet[3] = 0xFD;
    // keep unused channels values in the middle
    const uint16_t defaultValue = 50;

    dsm_controls_t p = { 50, 50, 50, 0, 0, 0 };

    uint16_t channelsVals[CHANNELS_NUMBER] = {
        p.pitch,
        p.roll,
        p.yaw,
        p.throttle,
        p.mode,
        p.rtl
    };

    xEventGroupSetBits(state_group, BIT_DSM_ENABLED);

    uint32_t bits = 0;

    TickType_t x22ms = pdMS_TO_TICKS(22);
    TickType_t tick = xTaskGetTickCount();

    for (;;) {
        vTaskDelayUntil(&tick, x22ms);

        if (xQueueReceive(dsm_controls_queue, &p, 0) == pdTRUE) {
            channelsVals[0] = p.pitch;
            channelsVals[1] = p.roll;
            channelsVals[2] = p.yaw;
            channelsVals[3] = p.throttle;
            channelsVals[4] = p.mode;
            channelsVals[5] = p.rtl;
        }

        bits = xEventGroupGetBits(state_group);

        if ((bits & BIT_NO_LINK) == BIT_NO_LINK) {
            channelsVals[0] = defaultValue;
            channelsVals[1] = defaultValue;
            channelsVals[2] = defaultValue;
            channelsVals[3] = 40;
            channelsVals[4] = p.mode;
            channelsVals[5] = p.rtl;
        }

        if ((bits & BIT_DSM_ENABLED) != BIT_DSM_ENABLED) {
            continue;
        }

        // channels and values
        for (uint8_t i = 0, j = 4; i < CHANNELS_NUMBER; ++i, j += 2) {
            uint16_t bytesPair = channelValueToBytes(i, scaleChannel(channelsVals[i]));
            dsm_packet[j] = ((uint8_t*)(&bytesPair))[1];
            dsm_packet[j + 1] = ((uint8_t*)(&bytesPair))[0];
        }

        DMA_Cmd(DMA1_Channel7, DISABLE);
        DMA1_Channel7->CNDTR = DSM_PACKET_SIZE;
        DMA_Cmd(DMA1_Channel7, ENABLE);

        xEventGroupClearBits(state_group, BIT_DSM_TASK);
    }
}
````

lobart, 2022-03-08:
@dagar 
what do you mean about initial byte?
For dsmx what is it?
It is first packet?

dagar, 2022-03-08:
From "Specification for Spektrum Remote Receiver Interfacing Enabling Use of Spektrum Remotes in Third-Party Products"
[image]

At one point the PX4 DSM parser tried to rely on this byte as it was being set correctly by legit Spektrum equipment, but ultimately we had to relax the parsing to tolerate unknown 3rd party hardware.

Could you give https://github.com/PX4/PX4-Autopilot/pull/18270 a try? If it's not working there we'll need to enable some debugging.

lobart, 2022-03-08:
Ok, I’ll try it tomorrow morning

lobart, 2022-03-08:
@dagar I use dsmx 22ms. dsm_packet[1]=0xa2

lobart, 2022-03-09:
@dagar I uploaded your firmware and not results. 
When i click Radio it is not sound [https://docs.px4.io/v1.9.0/en/getting_started/tunes.html#notify-positive-tone](https://docs.px4.io/v1.9.0/en/getting_started/tunes.html#notify-positive-tone)
And sticks don't move

lobart, 2022-03-09:
this structure of my packet
```c
    dsm_packet[0] = 0x00;
    dsm_packet[1] = 0xA2;
    dsm_packet[2] = 0x0B;
    dsm_packet[3] = 0xFD;
    channelsVals[0] = p.pitch;
    channelsVals[1] = p.roll;
    channelsVals[2] = p.yaw;
    channelsVals[3] = p.throttle;
    channelsVals[4] = p.mode;
    channelsVals[5] = p.rtl;
    dsm_packet[4] = bytesPair(channelsVals[0])[1]
    dsm_packet[5] = bytesPair(channelsVals[0])[0]
   ...
    dsm_packet[14] = bytesPair(channelsVals[5])[1]
    dsm_packet[15] = bytesPair(channelsVals[5])[0]
```

lobart, 2022-03-09:
I tried use dsm.h, dsm.cpp from 1.11.3, but there is no any results.
May be error is more global? not in dsm service?

lobart, 2022-03-09:
And i tried to use 1.12.3 with dsm.cpp, dsm.h from 1.11.2. And it was not worked.

closed 2025-10-30 by github-actions[bot]