PX4/PX4-Autopilot #19294: Dual MAG inconsistent compasses

farhangnaderi, 2022-03-08:
**Describe the bug**
I am trying to use Dual MAG on Pixhawk 5X and did calibration successfully. However I flew with Stabilized mode successfully but was not able to continue due to the error : "Compasses blah blah degrees inconsistent"

**To Reproduce**
Steps to reproduce the behavior:
1. Second compass to GPS2 
2. Enable the params accordingly
3. Calibrate compass
4. Arm the drone

**Expected behavior**
Successful Takeoff and Landing

**Log Files and Screenshots**
This has the whole procedure I followed:

https://review.px4.io/plot_app?log=957b7780-f202-4672-a383-c56619431752

These might be useful also:

https://review.px4.io/plot_app?log=3331be1a-bc7a-4ad0-9d89-979e8315cb02
https://review.px4.io/plot_app?log=8becd780-eac5-49cf-aebf-f0f736fa798d
https://review.px4.io/plot_app?log=8149baf0-3bb8-406e-ba2d-7857c1e2d302
https://review.px4.io/plot_app?log=92d65a13-161f-4bff-bf5b-b971c2b8c37b

Add screenshots to help explain your problem.

**Drone (please complete the following information):**
- X500 V2 with Pixhawk 5X 
- Third party GPS/GNSS module

**Additional context**
I need some guide on this if possible, please!

dagar, 2022-03-08:
> Got it. Maybe I can try CAL_MAG_PRIME then next.

That sensor either needs to be fixed or removed/disabled. Can you verify the wiring? 

Can you bootup and check the output of `dmesg`, `listener sensor_mag`, and `sensors status`?

farhangnaderi, 2022-03-24:
@dagar Conpass readings are nonsense. I will open another issue after further discussions with the manufacturer.

closed 2022-03-24 by farhangnaderi

aksjdakwhdijahskjdhkajbsf, 2024-04-02:
> The data for your 3rd mag (sensor_mag 2) is bad. [image]

How can i get the picture like this from my Log?