PX4/PX4-Autopilot #19322: QGroundControl loses connection to the GPS on the Cloudship simulation through Gazebo Software-in-the-Loop through PX4.

JamesAnawati, 2022-03-13:
QGroundControl loses connection to the GPS on the Cloudship simulation through Gazebo Software-in-the-Loop through PX4.

To Reproduce:
1. Boot Ubuntu 20.04
2. Boot QGroundControl (v4.1.6)
3. Open terminal and change directory to PX4 repository
         cd /path/to/PX4-Autopilot
4. Boot Cloundship simulation
         make px4_sitl gazebo_cloudship
5. Click through error message
6. Attempt a mission

Expected behavior:
Gazebo will open okay without issue, showing the cloudship wobbling. QGroundControl will connect to the simulation but return an error message saying that the firmware has a bug as it is missing the parameter: FW ASRP MODE. QGroundControl will then act mostly normal until GPS connection to the ship is lost and the clouship no longer responds to commands.

Screenshots
[image]
[image]

When I attempt the same procedure with any other vehicle type, there is no issue. Something is wrong with the cloudship simulation.

Thank you for your help.

Jaeyoung-Lim, 2022-03-14:
@JamesAnawati Although I find it quite hard to control the vehicle, I can't reproduce the gps loss issue you are reporting

JamesAnawati, 2022-03-14:
One of the ways I triggered the issue was by laying some waypoints in the mission planner, arming the vehicle, and selecting the mission mode. A few seconds later the issue arises. I am new to PX4 and autonomous vehicles in general so I was not sure if the difficulty to control the vehicle was my own fault. The vehicle also does not seen to follow any mission profile I give it.

Thank you for looking into this issue.