PX4/PX4-Autopilot #19333: ERROR | 'clock_settime' is unavailable: not available on iOS

omerts, 2022-03-15:
When trying to run on macOS Big Sure versino 11.4, getting the following error:
```
FAILED: platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o 
/usr/local/bin/ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"px4\" -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/Users/NothingButOne/develop/PX4-Autopilot/boards/px4/sitl/src -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/include -I/Users/NothingButOne/develop/PX4-Autopilot/build/px4_sitl_default -I/Users/NothingButOne/develop/PX4-Autopilot/build/px4_sitl_default/src/lib -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/common -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/common/include -I/Users/NothingButOne/develop/PX4-Autopilot/src -I/Users/NothingButOne/develop/PX4-Autopilot/src/include -I/Users/NothingButOne/develop/PX4-Autopilot/src/lib -I/Users/NothingButOne/develop/PX4-Autopilot/src/lib/matrix -I/Users/NothingButOne/develop/PX4-Autopilot/src/modules -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/include -I/Users/NothingButOne/develop/PX4-Autopilot/build/px4_sitl_default/external/Install/include -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/lockstep_scheduler/include -O2 -g -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator15.2.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-c99-designator -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-rtti -Wno-cast-align -std=gnu++14 -MD -MT platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o -MF platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o.d -o platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o -c /Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/drv_hrt.cpp
/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/drv_hrt.cpp:500:10: fatal error: 'clock_settime' is unavailable: not available on iOS
                return system_clock_settime(clk_id, ts);
                       ^

```

**To Reproduce**
Steps to reproduce the behavior:
1. clone the repository
2. cd into the cloned folder
3. run `make px4_sitl jmavsim`

**Expected behavior**
jmavsim should launch

Fuller log:
```
[23/530] Building CXX object platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o
FAILED: platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o 
/usr/local/bin/ccache /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"px4\" -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_POSIX -D__STDC_FORMAT_MACROS -Dnoreturn_function="__attribute__((noreturn))" -I/Users/NothingButOne/develop/PX4-Autopilot/boards/px4/sitl/src -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/include -I/Users/NothingButOne/develop/PX4-Autopilot/build/px4_sitl_default -I/Users/NothingButOne/develop/PX4-Autopilot/build/px4_sitl_default/src/lib -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/generic/generic/include -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/common -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/common/include -I/Users/NothingButOne/develop/PX4-Autopilot/src -I/Users/NothingButOne/develop/PX4-Autopilot/src/include -I/Users/NothingButOne/develop/PX4-Autopilot/src/lib -I/Users/NothingButOne/develop/PX4-Autopilot/src/lib/matrix -I/Users/NothingButOne/develop/PX4-Autopilot/src/modules -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/include -I/Users/NothingButOne/develop/PX4-Autopilot/build/px4_sitl_default/external/Install/include -I/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/lockstep_scheduler/include -O2 -g -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator15.2.sdk -fPIC -g -fdata-sections -ffunction-sections -fomit-frame-pointer -fmerge-all-constants -fno-signed-zeros -fno-trapping-math -freciprocal-math -fno-math-errno -fno-strict-aliasing -fvisibility=hidden -include visibility.h -Wall -Wextra -Werror -Warray-bounds -Wcast-align -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -fcolor-diagnostics -fdiagnostics-absolute-paths -Qunused-arguments -Wno-c99-designator -Wno-unknown-warning-option -Wno-unused-const-variable -Wno-varargs -fno-exceptions -fno-threadsafe-statics -Wreorder -Wno-overloaded-virtual -fno-rtti -Wno-cast-align -std=gnu++14 -MD -MT platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o -MF platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o.d -o platforms/posix/src/px4/common/CMakeFiles/px4_layer.dir/drv_hrt.cpp.o -c /Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/drv_hrt.cpp
/Users/NothingButOne/develop/PX4-Autopilot/platforms/posix/src/px4/common/drv_hrt.cpp:500:10: fatal error: 'clock_settime' is unavailable: not available on iOS
                return system_clock_settime(clk_id, ts);
                       ^
/Users/NothingButOne/develop/PX4-Autopilot/src/include/visibility.h:87:30: note: expanded from macro 'system_clock_settime'
#define system_clock_settime clock_settime
                             ^
/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator.sdk/usr/include/time.h:187:5: note: 'clock_settime' has been explicitly marked unavailable here
int clock_settime(clockid_t __clock_id, const struct timespec *__tp);
    ^
1 error generated.
[40/530] Building CXX object src/drivers/pwm_out_sim/CMakeFiles/drivers__pwm_out_sim.dir/PWMSim.cpp.o
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

**Additional context**
[image]

dagar, 2022-03-15:
It looks like it's trying to build for iphone?

```
 'clock_settime' is unavailable: not available on iOS
```

omerts, 2022-03-16:
@dagar that is what is happening automatically when running make px4_sitl jmavsim on macOS

dagar, 2022-03-16:
Do you have current PX4 master?

Did you run the macos setup script (`./Tools/setup/macos.sh`)? Does it complete successfully?

https://github.com/PX4/PX4-Autopilot/blob/master/Tools/setup/macos.sh

omerts, 2022-05-17:
@dagar  So sorry for late reply, I missed your response. 
Just ran macos.sh (didn't run it before because it was under Gazebo simulation), but unfortunately still the same error.