PX4/PX4-Autopilot #19338: Horizontal movement observed only with throttle inputs in ALTCTL mode

squizz617, 2022-03-15:
**Describe the bug**
Hi.
've come across a weird issue in ALTCTL mode. If I (1) send a remote control input (throttle: 0.8, RPY: 0) before arming, (2) switching to ALTCTL mode, (3) arm, and then (4) send consistent remote control inputs consisting of positive throttle value and zero roll, pitch, and yaw values, (e.g., throttle: 0.75, R/P/Y: 0), the drone somehow drifts horizontally. The [flight log](https://review.px4.io/plot_app?log=2c07f400-18b5-4898-8799-4778f11793e0) clearly shows that there were no RPY inputs, but still, non-zero vx and vy are observed.

**To Reproduce**
Steps to reproduce the behavior:
[terminal 1] make px4_sitl_rtps gazebo -j4
[terminal 2] Download and run the [PoC](https://gist.github.com/squizz617/dd73a2f9cb66a30a8367d44a24c78f6a)

**Expected behavior**
The drone should only move vertically.

**Log Files and Screenshots**
- Flight review: https://review.px4.io/plot_app?log=2c07f400-18b5-4898-8799-4778f11793e0

**Drone (please complete the following information):**
- Build: `px4_sitl_rtps`
- tested commit: `a2645418`

Thanks for looking into this!

Jaeyoung-Lim, 2022-03-15:
@squizz617 Since altitude control mode does not control position, the behavior seems reasonable.

Any imbalance of the hardware pr wind will lead to a drift of the vehicle. This is explained in the documentation: https://docs.px4.io/master/en/flight_modes/altitude_mc.html

squizz617, 2022-03-16:
Hi @Jaeyoung-Lim, thanks for the quick response.

> Any imbalance of the hardware pr wind will lead to a drift of the vehicle.

Given that the above situation happens in the **simulation**, where no external force or HW imbalance are expected, it doesn't seem reasonable to me.

Moreover, without the first dummy command in step (1) being sent before arming, which apparently should not take any effect on the later behavior, the drone moves only vertically, as expected. You can verify this by running the provided PoC. Am I missing anything here? Thanks!

Jaeyoung-Lim, 2022-03-16:
> Given that the above situation happens in the simulation, where no external force or HW imbalance are expected, it doesn't seem reasonable to me.

You can see the roll ground truth drifting in the log, while the setpoints are not moving. The gazebo simulation you are running is not a numerical model with rigid body assumptions and there are also different sensor biases being simulated. Therefore I would not say because it is a simulation, that everything should be in balance. 

 > Moreover, without the first dummy command in step (1) being sent before arming, which apparently should not take any effect on the later behavior, the drone moves only vertically, as expected. You can verify this by running the provided PoC. Am I missing anything here? Thanks!

What is important is that in altitude control mode the lateral position of the vehicle is **uncontrolled**. Is there a log without the dummy commands executed? I don't seem to see it in the issue

squizz617, 2022-03-16:
> The gazebo simulation you are running is not a numerical model with rigid body assumptions

Ah, I see. Thanks for the information!

I found in the flight log that without the first command, the drone actually failed to switch to Altitude mode and fell back to Position mode. That's why it was moving straight up. Closing!

closed 2022-03-16 by squizz617