PX4/PX4-Autopilot #19343: MAVLink serial port parameter documentation not being generated in some cases

hamishwillee, 2022-03-17:
Docs for parameters are autogenerated and pushed into the docs repo here: https://github.com/PX4/PX4-user_guide/blob/master/en/advanced_config/parameter_reference.md
This is getting regular updates, but some of the MAV_n_ params from https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/mavlink/module.yaml are not being generated in 
https://github.com/PX4/PX4-user_guide/blob/master/en/advanced_config/parameter_reference.md#mavlink

Specifically, these are appearing: MAV_n_CONFIG , MAV_n_FLOW_CTRL  , MAV_n_FORWARD, MAV_n_MODE , MAV_n_RADIO_CTL , MAV_n_RATE 

But these are not: MAV_n_BROADCAST, MAV_n_REMOTE_PRT, MAV_n_UDP_PRT

hamishwillee, 2022-03-17:
@bkueng @dagar The Ethernet config in 5x [rc.board_defaults](https://github.com/PX4/PX4-Autopilot/tree/master/boards/px4/fmu-v5x/init)/rc.board_defaults) sets up Ethernet to connect to the GCS.
It is obvious how you would modify this to stream to an offboard computer.

But how do you set it up to stream to both at the same time? There is no reason in theory you can't because they use different ports.

hamishwillee, 2022-03-17:
Further, can you tell me the difference between `MAV_2_REMOTE_PRT` and `MAV_2_UDP_PRT`? (these are the same value in  [rc.board_defaults](https://github.com/PX4/PX4-Autopilot/tree/master/boards/px4/fmu-v5x/init/rc.board_defaults) )

In simulation I understand we have both PX4 and QGC running on the same computer. The remote port is the port where PX4 broadcasts to - and QGC connects by listening to that port. There is another port on PX4 which PX4 listens to - so a remote system can initiate a connection if it wants.

[image]

Is that in any way related? If so, my concern is that PX4 and QGC are now on different computers. So I can see that in the above scenario how you would listen on the port on the same computer, but not how you would connect to the remote port - since you don't know what the IP address of the computer to send to ... unless this is all broadcast?

dagar, 2022-03-18:
To start with the problem with the documentation is that it's generated from px4_sitl_default and some of the parameters are board dependent (and it's going to get worse).

https://github.com/PX4/PX4-Autopilot/blob/9ab99a7689f79c3e0d2c344e33420dcde53c8cb7/Makefile#L350-L351

Switching it to px4_fmu-v5x_default would fix this particular problem (and a few others), but then we'd be missing things like simulation (SIM) parameters.

Obviously we'll need to do some a bit more deliberate for the documentation, maybe generating a crude mostly correct superset, or somehow showing specialization per board. 

Related to the MAV UDP parameters in particular, it would probably be more work, but updating posix builds (and simulator special cases) to primarily configure Mavlink via parameters would close some of these holes and in general be a better more representative experience in SITL (and actual Linux boards). https://github.com/PX4/PX4-Autopilot/blob/9ab99a7689f79c3e0d2c344e33420dcde53c8cb7/ROMFS/px4fmu_common/init.d-posix/px4-rc.mavlink#L4-L32