PX4/PX4-Autopilot #19344: ekf2 failed to use Vision for localization and PX4 keeps noticing: requesting home position

JaimeParker, 2022-03-17:
Hi, thank you for checking this, I'm working on using **ORB SLAM2** on my drone in the **gazebo** environment.
After I get everything done, it seems efk2 didn't get Vision information or failed to change to Vision mode, below are my steps:
### edit rcS
disable GPS localization and Barometer for height measurement, using Vision instead
```XML
	# GPS used
	#param set EKF2_AID_MASK 1
	# Vision used and GPS denied
	param set EKF2_AID_MASK 24

	# Barometer used for hight measurement
	#param set EKF2_HGT_MODE 0
	# Barometer denied and vision used for hight measurement
	param set EKF2_HGT_MODE 3
```
and importantly, delete parameters load in .ros:
```shell
rm ~/.ros/eeprom/parameters*
rm -rf ~/.ros/sitl*
```
so that the next time I launch the gazebo, rcS parameters will be loaded again.
### add a stereo camera in the launch file
I'm using `mavros_posix_sitl.launch`, and edit like this:
```xml
    <!-- vehicle model and world -->
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <!-- add stereo camera for iris -->
    <arg name="my_camera" default="iris_stereo_camera"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <!-- also need to revise sdf -->
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg my_camera)/$(arg my_camera).sdf"/>
```
### set .sh file
then I created a .sh file for ORB SLAM2, contents below:
```
rosrun ORB_SLAM2 Stereo 
~/catkin_ws/src/ORB_SLAM2/Vocabulary/ORBvoc.txt 
~/catkin_ws/src/ORB_SLAM2/Examples/Stereo/px4_sitl.yaml true
 /camera/left/image_raw:=/stereo/left/image_raw 
/camera/right/image_raw:=/stereo/right/image_raw 
/orbslam2/vision_pose/pose:=/mavros/vision_pose/pose
```
### Run
Finally, I started my trail
```shell
roslaunch px4 mavros_posix_sitl.launch
sh that.sh
```
I guess ORB is working well, I put a screenshot here, [link](https://images.gitee.com/uploads/images/2022/0317/170438_162b7dfb_8073751.png)
According to [others](https://www.yuque.com/xtdrone/manual_en/vslam), there should be information like below to show that Vision is functioning well, but it didn't appear on my screen.
```shell
INFO  [ecl/EKF] 23860000: reset position to ev position
INFO  [ecl/EKF] 23860000: commencing external vision position fusion
INFO  [ecl/EKF] 23860000: commencing external vision yaw fusion
```
and my PX4 keeps noticing me that hp: requesting home position, I can't use `commander takeoff` and I can't get into offboard mode, I'm really confused, can anyone help me?
here is a bug picture, [link](https://images.gitee.com/uploads/images/2022/0317/170707_806065bd_8073751.png)
```
[ INFO] [1647332081.616824996, 463.948000000]: HP: requesting home position
WARN  [commander] Failsafe enabled: no RC and no offboard
[ERROR] [1647332086.858252172, 466.871000000]: FCU: Failsafe enabled: no RC and no offboard
[ INFO] [1647332086.916993721, 466.923000000]: FCU: Failsafe mode activated
[ INFO] [1647332097.829447664, 473.031000000]: FCU: DISARMED by Auto disarm initiated
[ INFO] [1647332099.221400643, 473.948000000]: HP: requesting home position
INFO  [logger] closed logfile, bytes written: 560350
[ INFO] [1647332116.884205512, 483.948000000]: HP: requesting home position
[ INFO] [1647332134.370511761, 493.948000000]: HP: requesting home position
[ INFO] [1647332152.956565434, 503.948000000]: HP: requesting home position
[ INFO] [1647332168.367523905, 513.948000000]: HP: requesting home position
[ INFO] [1647332183.421351360, 523.949000000]: HP: requesting home position
[ INFO] [1647332198.407902287, 533.948000000]: HP: requesting home position
[ INFO] [1647332212.284682896, 543.948000000]: HP: requesting home position
```

Thanks in advance!

Jaeyoung-Lim, 2022-03-17:
@JaimeParker I see two problems

a) 
```
[ERROR] [1647332086.858252172, 466.871000000]: FCU: Failsafe enabled: no RC and no offboard
```
You have the RC failsafe enabled for offboard mode. This means that if you have no manual control input the vehicle will reject entering offboard mode. You can turn this off if you want, but this is a safety feature for real vehicles

b) If you are not sending a setpoint, you cannot get into offboard mode. Otherwise, it is unclear what the vehicle should do in offboard mode

JaimeParker, 2022-03-17:
> @JaimeParker I see two problems
>
> a)
>
> ```
> [ERROR] [1647332086.858252172, 466.871000000]: FCU: Failsafe enabled: no RC and no offboard
> ```
>
> You have the RC failsafe enabled for offboard mode. This [...]

@Jaeyoung-Lim , thanks for your reply, sir.
a) that's right, after I started QGroundControl, opening virtual joystick, this error is fixed I guess.
b) thanks for what  you mentioned, I just realized that we need to send some points in some rate before entering into offborad mode, which can be referred here: https://docs.px4.io/master/en/ros/mavros_offboard.html
However, I still have no info said that Vision is in use and can't use `commander takeoff`, still no answer.
Again, thank you, sir.

JaimeParker, 2022-03-24:
fixed, need to revise `ORB_SLAM2/Example/ROS/ORB_SLAM2/src/ros_stereo.cc`, link here: https://gitee.com/hazyparker/my-research/blob/master/3_Onboard_SLAM/XTDrone_Method/ros_stereo.cc

mengchaoheng, 2023-04-21:
The same problem

closed 2025-10-29 by github-actions[bot]