PX4/PX4-Autopilot #19345: EKF2 not executing when using takeoff command

stevedanomodolor, 2022-03-17:
I am using rox foxy on an ubuntu 20.4 machine
When I call this command `commander takeoff`, I get the following but the drone does not move atall 

```
pxh> INFO  [tone_alarm] home set
pxh> cINFO  [tone_alarm] notify negative
pxh> commander takeoff
pxh> INFO  [commander] Armed by internal command	
INFO  [navigator] Using minimum takeoff altitude: 2.50 m	
INFO  [commander] Takeoff detected	
WARN  [commander] Failsafe enabled: no datalink	
INFO  [commander] Failsafe mode activated	
INFO  [tone_alarm] battery warning (fast)
INFO  [navigator] RTL HOME activated	
INFO  [navigator] RTL: landing at home position.	
INFO  [navigator] RTL: climb to 489 m (1 m above destination)	
INFO  [commander] Failsafe mode deactivated	
INFO  [navigator] RTL: return at 489 m (1 m above destination)	
INFO  [navigator] RTL: land at destination	
INFO  [commander] Landing detected	
INFO  [commander] Disarmed by landing	
INFO  [logger] closed logfile, bytes written: 3975567
```

It seems that the EKF did not initialize. When running the `make px4_sitl_rtps gazebo`, the output is ass follows: 

```
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/stevedan/project_001/navity/control_mo.../mission_controller/PX4-Autopilot/build/px4_sitl_rtps
SITL ARGS
sitl_bin: /home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot
build_path: /home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps/build_gazebo
GAZEBO_MODEL_PATH /home/stevedan/gazebo_models/models::/home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/stevedan/project_001/navity/navity_ws/install/px4_ros_com/lib:/home/stevedan/project_001/navity/navity_ws/install/px4_msgs/lib:/home/stevedan/ros2_foxy/ros2-linux/opt/yaml_cpp_vendor/lib:/home/stevedan/ros2_foxy/ros2-linux/opt/rviz_ogre_vendor/lib:/home/stevedan/ros2_foxy/ros2-linux/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/opt/ros/noetic/lib:/home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps/build_gazebo
empty world, setting empty.world as default
Using: /home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/Tools/sitl_gazebo/models/iris/iris.sdf
gzserver not ready yet, trying again!
SITL COMMAND: "/home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps/bin/px4" "/home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps"/etc -s etc/init.d-posix/rcS -t "/home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot"/test_data
Creating symlink /home/stevedan/project_001/navity/control_module/mission_controller/PX4-Autopilot/build/px4_sitl_rtps/

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
INFO  [parameters] BSON document size 333 bytes, decoded 333 bytes (INT32:14, FLOAT:3)
[param] Loaded: eeprom/parameters_10016
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [micrortps_client] UDP transport: ip address: 127.0.0.1; recv port: 2019; send port: 2020
INFO  [micrortps_client] UDP transport: Trying to connect...
INFO  [micrortps_client] UDP transport: Connected to server!
[... 168 lines cut ...]
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_odometry advertise failed (5)
ERROR [uORB] estimator_wind advertise failed (5)
ERROR [uORB] ekf2_timestamps advertise failed (5)
ERROR [uORB] estimator_baro_bias advertise failed (5)
ERROR [uORB] estimator_event_flags advertise failed (5)
ERROR [uORB] estimator_gps_status advertise failed (5)
ERROR [uORB] estimator_innovation_test_ratios advertise failed (5)
ERROR [uORB] estimator_innovation_variances advertise failed (5)
ERROR [uORB] estimator_innovations advertise failed (5)
ERROR [uORB] estimator_optical_flow_vel advertise failed (5)
ERROR [uORB] estimator_sensor_bias advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_status_flags advertise failed (5)
ERROR [uORB] estimator_status advertise failed (5)
ERROR [uORB] estimator_visual_odometry_aligned advertise failed (5)
ERROR [uORB] yaw_estimator_status advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [ekf2] publication instance problem: -1 att: -1 lpos: -1 gpos: -1
ERROR [ekf2] alloc and init failed imu: 0 mag:1

```

bresch, 2022-03-18:
> When I call this command commander takeoff, I get the following but the drone does not move at all

The shell tells you that:
```
WARN  [commander] Failsafe enabled: no datalink	
INFO  [commander] Failsafe mode activated
```

Meaning that the drone goes into failsafe because it doesn't have a connection to a data link (i.e.: ground station), meaning that you need to open QGC. If you don't want to have QGC open, you can also deactivate the datalink failsafe action.

stevedanomodolor, 2022-03-18:
how do i do that, i ran the `param set NAV_DLL_ACT 0` but still did work desable

stevedanomodolor, 2022-03-18:
I keep getting this when I start to restart the simulation 

ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_odometry advertise failed (5)
ERROR [uORB] estimator_wind advertise failed (5)
ERROR [uORB] ekf2_timestamps advertise failed (5)
ERROR [uORB] estimator_baro_bias advertise failed (5)
ERROR [uORB] estimator_event_flags advertise failed (5)
ERROR [uORB] estimator_gps_status advertise failed (5)
ERROR [uORB] estimator_innovation_test_ratios advertise failed (5)
ERROR [uORB] estimator_innovation_variances advertise failed (5)
ERROR [uORB] estimator_innovations advertise failed (5)
ERROR [uORB] estimator_optical_flow_vel advertise failed (5)
ERROR [uORB] estimator_sensor_bias advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_status_flags advertise failed (5)
ERROR [uORB] estimator_status advertise failed (5)
ERROR [uORB] estimator_visual_odometry_aligned advertise failed (5)
ERROR [uORB] yaw_estimator_status advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [ekf2] publication instance problem: -1 att: -1 lpos: -1 gpos: -1
ERROR [ekf2] alloc and init failed imu: 1 mag:1
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_odometry advertise failed (5)
ERROR [uORB] estimator_wind advertise failed (5)
ERROR [uORB] ekf2_timestamps advertise failed (5)
ERROR [uORB] estimator_baro_bias advertise failed (5)
ERROR [uORB] estimator_event_flags advertise failed (5)
ERROR [uORB] estimator_gps_status advertise failed (5)
ERROR [uORB] estimator_innovation_test_ratios advertise failed (5)
ERROR [uORB] estimator_innovation_variances advertise failed (5)
ERROR [uORB] estimator_innovations advertise failed (5)
ERROR [uORB] estimator_optical_flow_vel advertise failed (5)
ERROR [uORB] estimator_sensor_bias advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_status_flags advertise failed (5)
ERROR [uORB] estimator_status advertise failed (5)
ERROR [uORB] estimator_visual_odometry_aligned advertise failed (5)
ERROR [uORB] yaw_estimator_status advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [ekf2] publication instance problem: -1 att: -1 lpos: -1 gpos: -1
ERROR [ekf2] alloc and init failed imu: 1 mag:1
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_odometry advertise failed (5)
ERROR [uORB] estimator_wind advertise failed (5)
ERROR [uORB] ekf2_timestamps advertise failed (5)
ERROR [uORB] estimator_baro_bias advertise failed (5)
ERROR [uORB] estimator_event_flags advertise failed (5)
ERROR [uORB] estimator_gps_status advertise failed (5)
ERROR [uORB] estimator_innovation_test_ratios advertise failed (5)
ERROR [uORB] estimator_innovation_variances advertise failed (5)
ERROR [uORB] estimator_innovations advertise failed (5)
ERROR [uORB] estimator_optical_flow_vel advertise failed (5)
ERROR [uORB] estimator_sensor_bias advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_status_flags advertise failed (5)
ERROR [uORB] estimator_status advertise failed (5)
ERROR [uORB] estimator_visual_odometry_aligned advertise failed (5)
ERROR [uORB] yaw_estimator_status advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
[... 159 lines cut ...]
ERROR [uORB] estimator_wind advertise failed (5)
ERROR [uORB] ekf2_timestamps advertise failed (5)
ERROR [uORB] estimator_baro_bias advertise failed (5)
ERROR [uORB] estimator_event_flags advertise failed (5)
ERROR [uORB] estimator_gps_status advertise failed (5)
ERROR [uORB] estimator_innovation_test_ratios advertise failed (5)
ERROR [uORB] estimator_innovation_variances advertise failed (5)
ERROR [uORB] estimator_innovations advertise failed (5)
ERROR [uORB] estimator_optical_flow_vel advertise failed (5)
ERROR [uORB] estimator_sensor_bias advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_status_flags advertise failed (5)
ERROR [uORB] estimator_status advertise failed (5)
ERROR [uORB] estimator_visual_odometry_aligned advertise failed (5)
ERROR [uORB] yaw_estimator_status advertise failed (5)
ERROR [uORB] estimator_states advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [ekf2] publication instance problem: -1 att: -1 lpos: -1 gpos: -1
ERROR [ekf2] alloc and init failed imu: 1 mag:1
ERROR [uORB] estimator_attitude advertise failed (5)
ERROR [uORB] estimator_local_position advertise failed (5)
ERROR [uORB] estimator_global_position advertise failed (5)
ERROR [uORB] estimator_odometry advertise failed (5)

closed 2022-03-18 by stevedanomodolor