PX4/PX4-Autopilot #19348: [PreFlightCheck] Mode not suitable for takeoff

julianoes, 2022-03-17:
**Describe the bug**
The simple MAVSDK takeoff and land example does not work properly anymore, that is, it only works once, then it refuses to arm saying "Mode not suitable for takeoff" as it is in Land mode.
This used to work and is very confusing.

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL
2. Try https://github.com/mavlink/MAVSDK-Python/blob/main/examples/takeoff_and_land.py with latest mavsdk from pip installed.
3. Run it once.
4. Run it again.

julianoes, 2022-03-20:
@dagar @MaEtUgR ?

Czw007, 2022-04-12:
How to modify the code of "takeoff_and_land" to avoid "Mode not suitable for takeoff"?

> **Describe the bug** The simple MAVSDK takeoff and land example does not work properly anymore, that is, it only works once, then it refuses to arm saying "Mode not suitable for takeoff" as it is in [...]

Hello! Can you tell me how to modify the code of "takeoff_and_land.py" to avoid this error "Mode not suitable for takeoff"

MaEtUgR, 2022-04-13:
@julianoes @Czw007 Does https://github.com/PX4/PX4-Autopilot/pull/19449 not hotfix/unblock this issue? At least for the case you're in land mode after landing. We need an automatic switch to keep being compatible to the arm and then switch mode workflow. I still suggest changing the workflow to first switch the mode and then arm in the future or possibly making it atomic in the state machine in further future.

julianoes, 2022-04-13:
Yes, thanks @MaEtUgR

closed 2022-04-13 by julianoes