PX4/PX4-Autopilot #19349: MAVSDK offboard examples not working without manual control present

julianoes, 2022-03-18:
**Describe the bug**
The MAVSDK offboard examples stopped working. It fails to takeoff in offboard mode even with `NAV_RCL_ACT` set to `0` claiming:
> `WARN  [commander] Failsafe enabled: No manual control stick input`

**To Reproduce**
Steps to reproduce the behavior:
1. Start SITL
2. Set `NAV_RCL_ACT` to `0`
3. Run MAVSDK example: e.g. https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_position_ned.py

I get:
```
pxh> 
pxh> 
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative
INFO  [commander] Armed by external command	
WARN  [commander] Failsafe enabled: No manual control stick input	
INFO  [commander] Failsafe mode activated	
INFO  [tone_alarm] battery warning (fast)
INFO  [commander] Disarmed by auto preflight disarming	
INFO  [logger] closed logfile, bytes written: 3283058
```

benjinne, 2022-03-18:
Is `COM_RCL_EXCEPT` set to ignore RC in offboard? I made some changes to how the offboard failsafes work in this MR https://github.com/PX4/PX4-Autopilot/pull/18160 and could be related.

omerts, 2022-04-11:
@benjinne I running into the same issue, can't go into offboard mode. Tried playing around with `COM_RCL_EXCEPT`, but it didn't help. This is really a show stopper bug, as you can't use offboard mode.

omerts, 2022-04-11:
The following param settings fixed it for me:

NAV_DLL_ACT = 0
NAV_RCL_ACT = 0
COM_RCL_EXCEPT = 4

Taken from: https://discuss.px4.io/t/offboard-mode-in-sitl/25727/3

julianoes, 2022-04-13:
Thanks @omerts. We should probably set that param in the example, or at least make a comment.

julianoes, 2022-06-23:
I'm currently trying to fix this for MAVSDK tests: https://github.com/mavlink/MAVSDK/pull/1813

julianoes, 2022-09-03:
@MaEtUgR sorry, this fell of my radar. You fixed this in the meantime, right?

fabrizioschiano, 2022-11-24:
@MaEtUgR Please check if this is somehow related to your problem: https://github.com/PX4/PX4-Autopilot/issues/20667#issuecomment-1326648176

ButyAfter, 2025-09-16:
> 以下参数设置为我解决了这个问题：
> 
> NAV_DLL_ACT = 0 NAV_RCL_ACT = 0 COM_RCL_EXCEPT = 4
> 
> 摘自：https://discuss.px4.io/t/offboard-mode-in-sitl/25727/3

我也遇到了这个问题，但是我按你说的这个三个参数设置了还是报错：commander] Disarmed by auto preflight disarming

[image]

[image]