PX4/PX4-Autopilot #19356: Follow me Mode

gendebeat, 2022-03-18:
hello.
i see that are a "proposal for new follow-me mode".
but im using 1.12.3 version and i try follow-me and drone dont do yaw to face me. when it followme, the yaw was sometimes 
fixed (meybe north) or erratic. 
Then i do SITL for same version and i see that the drone dont face to me, is this correct? 
In sitl the yaw are more fixed because is simulation, i think
I see videos from youtube with px4 1.5  and the dron is facing (or camera gimbal...) 
https://www.youtube.com/watch?v=B1fVHaogvHc

I do SITL of version 1.10  and same, dont face the remote gps
In what stage  is follow-me mode today, or its broken since 1.5....? or what are i missing..?
thanks

gendebeat, 2022-03-18:
owww ,sorry, i see de attached document now of "proposal for new follow-me mode"
https://docs.google.com/document/d/1E04Rn6sxYNqKve4-KZYdC2-9LExKmHKuYjWZlwf61rA/edit
that says 
Yaw tracking broken : The quadcopter doesn’t face the user, it’s pointing in a wrong direction! 
and other bad things
before 1.5 version it was broken, involution of this part of the software
otherwise thanks to all people involved in Px4

MaEtUgR, 2022-06-03:
There's a brand new follow me implementation under way but just not merged yet on master:
https://github.com/PX4/PX4-Autopilot/pull/18026
I suggest you give that one a try and give feedback.

junwoo091400, 2022-06-07:
Hi @gendebeat, as you mentioned the current follow me has the yaw problem, and it should get fixed by the improved follow me PR. I will update you once we get it merged!

junwoo091400, 2022-12-02:
@gendebeat it was merged! And you can watch a presentation about it in detail now: https://youtu.be/-1MASuJDGn4

closed 2022-12-02 by junwoo091400