PX4/PX4-Autopilot #19360: Problem when running "roslaunch px4 mavros_posix_sitl.launch" in Ubuntu20.04

AOA81, 2022-03-20:
When I trying to run `roslaunch px4 mavros_posix_sitl.launch`, I can launch the Gazebo11 and load the model "solo" or "iris" successfully, but the propellers of the quardrotor rotate automatically without any command from me, besides, I can't connect the QGC or ROS. 
It seems that Gazebo failed to communicate with the px4 through TCP 4560 because I can find `[simulator] Waiting for simulator to accept connection on TCP port 4560` in the terminal but can't find `INFO  [simulator] Simulator connected on TCP port 4560.`
Besides, there is no problem when running `make px4_sitl gazebo` or `make px4_sitl gazebo_solo`, everything goes OK, thus I think it's not due to the environments of PX4, Gazebo or MAVROS which were all installed under the guidance of the official site.

How could I fix this problem?

Thanks!

Some related information is shown here.

> NODES
>   /
>     gazebo (gazebo_ros/gzserver)
>     gazebo_gui (gazebo_ros/gzclient)
>     mavros (mavros/mavros_node)
>     sitl (px4/px4)
>     vehicle_spawn_A_O_A_32671_6861078203187157316 (gazebo_ros/spawn_model)
> 
> auto-starting new master
> process[master]: started with pid [32690]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 16786b36-a83e-11ec-8d75-273b89124606
> process[rosout-1]: started with pid [32711]
> started core service [/rosout]
> process[mavros-2]: started with pid [32719]
> [ INFO] [1647774516.655297518]: FCU URL: udp://:14540@localhost:14557
> [ INFO] [1647774516.656837619]: udp0: Bind address: 0.0.0.0:14540
> [ INFO] [1647774516.656978123]: udp0: Remote address: 127.0.0.1:14557
> [ INFO] [1647774516.657010837]: GCS bridge disabled
> [ INFO] [1647774516.663608231]: Plugin 3dr_radio loaded
> [ INFO] [1647774516.664546947]: Plugin 3dr_radio initialized
> [ INFO] [1647774516.664580289]: Plugin actuator_control loaded
> [ INFO] [1647774516.666316224]: Plugin actuator_control initialized
> ........
> [ INFO] [1647774516.679047636]: Plugin esc_status loaded
> [ INFO] [1647774516.679609856]: Plugin esc_status initialized
> [ INFO] [1647774516.679649369]: Plugin esc_telemetry loaded
> [ INFO] [1647774516.679972277]: Plugin esc_telemetry initialized
> [ INFO] [1647774516.680068395]: Plugin fake_gps loaded
> process[sitl-3]: started with pid [32726]
> INFO  [px4] Creating symlink /home/aoa/Documents/PX4-Autopilot/build/px4_sitl_default/etc -> /home/aoa/.ros/etc
> 
> ______  __   __    ___ 
> | ___ \ \ \ / /   /   |
> | |_/ /  \ V /   / /| |
> |  __/   /   \  / /_| |
> | |     / /^\ \ \___  |
> \_|     \/   \/     |_/
> 
> px4 starting.
> 
> INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
> process[gazebo-4]: started with pid [32727]
> Info: found model autostart file as SYS_AUTOSTART=1014
> [ INFO] [1647774516.690276995]: Plugin fake_gps initialized
> [ INFO] [1647774516.690384445]: Plugin ftp loaded
> process[gazebo_gui-5]: started with pid [32747]
> INFO  [param] selected parameter default file eeprom/parameters_1014
> [ INFO] [1647774516.694757002]: Plugin ftp initialized
> [ INFO] [1647774516.694879240]: Plugin geofence loaded
> process[vehicle_spawn_A_O_A_32671_6861078203187157316-6]: started with pid [32755]
> INFO  [parameters] BSON document size 406 bytes, decoded 406 bytes
> [ INFO] [1647774516.697053970]: Plugin geofence initialized
> [ INFO] [1647774516.697127833]: Plugin global_position loaded
> [param] Loaded: eeprom/parameters_1014
> [ INFO] [1647774516.705827771]: Plugin global_position initialized
> [ INFO] [1647774516.705938736]: Plugin gps_input loaded
> [ INFO] [1647774516.707539751]: Plugin gps_input initialized
> [ INFO] [1647774516.707619745]: Plugin gps_rtk loaded
> [ INFO] [1647774516.709201361]: Plugin gps_rtk initialized
> [ INFO] [1647774516.709298253]: Plugin gps_status loaded
> ........
> [ INFO] [1647774516.846446533]: Plugin waypoint loaded
> [ INFO] [1647774516.850240255]: Plugin waypoint initialized
> [ INFO] [1647774516.850277014]: Plugin wheel_odometry blacklisted
> [ INFO] [1647774516.850352660]: Plugin wind_estimation loaded
> [ INFO] [1647774516.850709171]: Plugin wind_estimation initialized
> [ INFO] [1647774516.850776020]: Autostarting mavlink via USB on PX4
> [ INFO] [1647774516.850806266]: Built-in SIMD instructions: SSE, SSE2
> [ INFO] [1647774516.850827003]: Built-in MAVLink package version: 2022.2.2
> [ INFO] [1647774516.850851771]: Known MAVLink dialects: common ardupilotmega ASLUAV AVSSUAS all development icarous matrixpilot paparazzi standard storm32 uAvionix ualberta
> [ INFO] [1647774516.850872554]: MAVROS started. MY ID 1.240, TARGET ID 1.1
> INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
> PX4 SIM HOST: localhost
> INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
> [INFO] [1647774517.060357, 0.000000]: Loading model XML from file /home/aoa/Documents/PX4-Autopilot/Tools/sitl_gazebo/models/solo/solo.sdf
> [INFO] [1647774517.062827, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
> [ INFO] [1647774517.166791216]: Finished loading Gazebo ROS API Plugin.
> [ INFO] [1647774517.167675034]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
> [ INFO] [1647774517.255758066]: Finished loading Gazebo ROS API Plugin.
> [ INFO] [1647774517.256664993]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
> [ INFO] [1647774518.187283669]: waitForService: Service [/gazebo/set_physics_properties] is now available.
> [ INFO] [1647774518.203194920]: Physics dynamic reconfigure ready.
> [INFO] [1647774518.268388, 0.064000]: Calling service /gazebo/spawn_sdf_model
> 
> libcurl: (5) Could not resolve proxy: https
> [INFO] [1647774518.526605, 0.168000]: Spawn status: SpawnModel: Successfully spawned entity
> [vehicle_spawn_A_O_A_32671_6861078203187157316-6] process has finished cleanly
> log file: /home/aoa/.ros/log/16786b36-a83e-11ec-8d75-273b89124606/vehicle_
spawn_A_O_A_32671_6861078203187157316-6*.log
[image]

AOA81, 2022-03-21:
Thanks for your focus and I've solved this problem by fixing the ~/.bashrc file which load the `GAZEBO_PLUGIN/MODEL_PATH` when open a terminal. 
This problem is result from the incorrect `GAZEBO_PLUGIN_PATH`, which means Gazebo failed to load related plugins built by px4 like `libgazebo_imu_plugin.so` and `libgazebo_gps_plugin.so` while simulating. The error massage can be found in `~/.gazebo/server-11345/default.log` instead of the terminal when running roslaunch.
The correct path should be like:
```
#source $(pwd)/Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
#export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
#export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
```
where $(pwd) denotes the home directory of PX4-Autopilot.

closed 2022-03-21 by AOA81

aoberai, 2022-12-17:
Nice find, saving this for future reference. Had the same issue after the recent refactor of setup_gazebo.bash and didn't realize because of the nondescript error message