PX4/PX4-Autopilot #19361: Unclear intention of sign function

tstastny, 2022-03-21:
**Describe the bug**
https://github.com/PX4/PX4-Autopilot/blob/c86d5769eaa719d5a3898060bf81b074d36228be/src/lib/matrix/matrix/helper_functions.hpp#L155-L159

Not sure if bug - but no documentation on intention so wanted to clarify. If one runs the following on the matrix lib sign function:
```
#include <iostream>
#include <cfloat>

float signOf(float val) {
    return float((FLT_EPSILON < val) - (val < FLT_EPSILON));
}

int main() {

    std::cout << signOf(-1.0f) << std::endl;
    std::cout << signOf(-FLT_EPSILON) << std::endl;
    std::cout << signOf(0.0f) << std::endl;
    std::cout << signOf(FLT_EPSILON) << std::endl;
    std::cout << signOf(1.0f) << std::endl;

    return 0;
}
```
the output is:
```
-1
-1
-1
0
1
```
**Expected behavior**
`sign` functions either have a positive or negative definition for a value input equal to zero (so that means the only two outputs possible are -1 or 1), **or** they have a 0 criteria, either exactly =0, or a deadzone.
1. is it intended that the sign function should return zero if within a "deadzone"?
2. if so, shouldn't 0 also return 0? e.g. `(T(FLT_EPSILON) < val) - (val < -T(FLT_EPSILON))`

Regardless of the intention, after this discussion we should update the method documentation.

MaEtUgR, 2022-03-21:
> is it intended that the sign function should return zero if within a "deadzone"?

It is, for the definition that cannot return zero there's the function `signNoZero()`:
https://github.com/PX4/PX4-Autopilot/blob/c86d5769eaa719d5a3898060bf81b074d36228be/src/lib/mathlib/math/Functions.hpp#L50-L55

They are in different places because of the historical dependencies of the matrix and ecl submodule dependencies. I'd vote for putting them in the same place outside of the matrix library and making the matrix library depend on e.g. math::functions explicitly in CMake.

> if so, shouldn't 0 also return 0?

Totally, let me check 👀 

> we should update the method documentation

I completely agree, I can add that.

MaEtUgR, 2022-03-21:
@tstastny I suggest this https://github.com/PX4/PX4-Autopilot/pull/19366 and then I'll make a pr to move the functions.

PR #19366 (2022-03-23): Test and fix sign function
**Describe problem solved by this pull request**
Fixes #19361 thanks to @tstastny !
- The sign function has no documentation
- There are no unit tests for the sign function
- The sign function has a bug where unexpectedly sign(0) = -1 and sign(FLT_EPSILON) = 0

**Describe your solution**
Add documentation and unit tests.
For the bug, I'm really not sure why the putative deadzone was added here: https://github.com/PX4/PX4-Autopilot/pull/12905/files#diff-b76fc4ece0dac2e83a0e972fbea85372175487ed12f9146ee211252e51da3494R51
It never worked before. It would need to be `(T( [... cut]
closed 2022-03-23 by MaEtUgR