PX4/PX4-Autopilot #19363: Micro RTPS agent not working on USB

karanjotsingh010, 2022-03-21:
Hi
I am working on Pixhawk 5x and trying to setup microRTPS agent between the Pixhawk and ROS2 foxy on my laptop. My laptop runs on Ubuntu 20.04. The Pixhawk is connected to the laptop via USB on the port /dev/ttyACM0. Of course, px4_ros_com_ros2 is built and sourced before running the agent. I have enabled read and write on the USB port using the command "sudo chmod 777 /dev/ttyACM0". As i try to establish the connection with Pixhawk using microRTPS agent, I get the following error.

[image]

I have enabled the microRTPS client for Pixhawk on Mavlink console through QGS:

[image]

I can see the data being transmitted in QGS through Mavlink inspector which means the data transmission is working:
[image]

Just need to access this data in a listener in ROS2. Lack of documentation on the ROS2-PX4 bridge for the Pixhawk hardware is a big issue in this case. Any solution for this issue with microRTPS agent connection for ROS2 and Pixhawk 5x??

benjinne, 2022-03-21:
I've used USB but I think it a different setup. I hookup to one of the telem ports with an FTDI adapter and it'll talk to the micrortps_agent no problem. I just set the `RTPS_CONFIG` to the correct port and point the agent to my FTDI adapter and it works

karanjotsingh010, 2022-03-22:
@benjinne Thanks for the prompt reply. I can see the options for RTPS_config are Telem, GPS or Pixhawk Payload Bus. I am trying to use either USB or UDP (Ethernet) port to set up the RTPS.  So I assume Pixhawk Payload Bus is the option for that. As an alternate, is it possible for you to share the pic of your setup?? I can switch to FTDI cable if my plan does not work out...
Also did you use the default "sensor_combined_listener.cpp" code available in px4_ros_comm examples (listener) for subscribing to the sensor_combined topic? Or did you make any changes?

karanjotsingh010, 2022-03-22:
Got the connection to work.
For reference, following are the steps: 
1. Connect Ethernet port of Pixhawk 5x to ethernet port of companion computer.
2. Set the IP address to static IP for the wired connection on the companion computer. For my purpose, I use the static IP address of 192.168.0.4 with a subnet mask of 255.255.255.0 on the companion computer.
3. Connect the Pixhawk to a system with QGC installed. Set RTPS_config to "Pixhawk Payload Bus".
4. For stable connection, disable MAVlink from in the parameter list: disable "MAV_0_config, MAV_1_config and MAV_2_config".
5. Enable RTPS client from MAVlink console in QGC using the following command: micrortps_client start -t UDP -i 192.168.0.4, where 192.168.0.4 is the static IP address of companion computer in my case.
6. Enable RTPS agent in the companion computer using the following command: micrortps_agent start -t UDP. 
7. Use the "sensor_combined_listener.cpp" code in the folder px4_ros_com_ros2/src/px4_ros_com/src/examples/listeners to check data reception.

Thanks a lot @benjinne for your help...

kang-hyeok, 2022-04-01:
Hi, I'm currently trying to perform FastRTPS communication between 5X ↔ Jetson NX.
I'd like to inquire about two things.

1. Could you tell me the PX4 firmware version of the 5X you used?
2. Could you tell me how you set up the microtps agent in ubntu 20.04?
   - I also set up and run libraries such as ros2 and px4_ros_com_ros2 related to microrts agent to communicate with 5x, 
     but an error occurred as shown in the picture below.

[image]
 
     Did you happen to have the same error?

karanjotsingh010, 2022-04-01:
Hey Kang-hyeok
The error you are getting is probably because you are using an older version of px4. If you are using Pixhawk 5x, just download the latest version of px4_fmuv5x.
The main source of your error is the mismatch of messages in px4_ros_com_ros2 and PX4-Autopilot folders.
You can find the message list in "urtps_bridge_topics.yaml" in the folder ~/PX4-Autopilot/msg/tools. Open new terminal and enter: cd ~/PX4-Autopilot/msg/tools to migrate to this folder. Then you can run this code to generate the same list of messages in px4_ros_com_ros2 folder: python3 uorb_to_ros_urtps_topics.py -i urtps_bridge_topics.yaml -o ~/px4_ros_com_ros2/src/px4_ros_com/templates/urtps_bridge_topics.yaml. That should remove the mismatch in message ID when you build px4_ros_com_ros2. 
Additionally, in order to build px4_ros_com_ros2, clean all bash first. Which can be done by opening new terminal and enter: cd ~/px4_ros_com_ros2/src/px4_ros_com/scripts. Then enter: source clean_all.bash.

Hope it helps.

karanjotsingh010, 2022-04-01:
Nopes. You should be able to connect to QGC. You have to connect your pixhawk 5x to your laptop. The command to build PX4-Autopilot is: make px4_fmuv5x-rtps. When you build it, upload firmware using QGC.

kh-0306, 2022-04-04:
As you said, I built it from master version with the make_fmu-v5x_rtps command and uploaded it from qgc to 5x, but if the upload message appears successfully, I still cannot connect to qgc. Do you know the reason for this?

karanjotsingh010, 2022-04-04:
Sorry I have not faced this problem!!

gsoga, 2022-04-12:
Hi @kh-0306, I have the same problem

In my case, I upload the firmware (built with make fmu-v5x_rtps) and I set the parameter RTPS_CONFIG to TELEM2. When reboot, after a while, QGC (v4.1.7) show this message:
[image]

I have checked micrortps_agent and I am not getting messages from TELEM2

Did you find a solution?

kh-0306, 2022-04-12:
When I tested with tel1~3, gps2 port, I didn't get rtps message from nx with tel2 port. I haven't solved the problem yet :(

karanjotsingh010, 2022-04-12:
Guys as far as i understand, messages to qgc are not read through telem port. They are enabled through mavlink tab in the parameters list and you have to connect through usb port. It is quite straightforward.
As far as RTPS goes, you have to enable RTPS config either on udp (ethernet port) or telem 2. If you use telem 2, then you need to use ftdi cable to connect to your system or controller and enable micro rtps agent on the controller side. Also nable micro rtps client on the qgc.

gsoga, 2022-04-12:
Thank you both for your response.
I use qgc to load firmware and configure the 5X (with USB port), but to communicate with ROS2 I use ftdi cable, as you mentioned, through TELEM2 port. I have a cube orange working and I am trying the 5X.
The problem is when setting the CONFIG_RTPS variable on a port, it doesn't matter which one, including the Pixhawk Payload Bus (working for you), after rebooting the 5X, qgc can't load the parameters, only allows reloading a firmware

I've checked the latest commits and undone some changes:

- I used the original file [boards/px4/fmu-v5x/rtps.px4board ](https://github.com/PX4/PX4-Autopilot/blob/f91aa76645093e2402d3bdb6f70072ca0608c7b8/boards/px4/fmu-v5x/rtps.px4board)
- I used the file before the last commit [src/modules/micrortps_bridge/CMakeLists.txt](https://github.com/PX4/PX4-Autopilot/blob/0885f3766708e3f5bbd2b99ebc3b17247a4f2b23/src/modules/micrortps_bridge/CMakeLists.txt)

After that, I rebuilt fmu-v5x_rtps and now I can set CONFIG_RTPS param correctly and comunicate with ROS2 through TELEM2 with ftdi cable

Maybe the problem is related to disabling some SYSTEMCMDS parameters, appreciated if anyone can shed some more light

karanjotsingh010, 2022-04-12:
I think the problem will be clearer if you can share a pic of the setup. I have never faced these problems but somebody else might be able to help...

dsix-ls2n, 2022-04-13:
Hello,
I just found that I was unable to run the mirco_rtps agent (hardware crash) on Pixhawk5x with the latest master version. It was working fine with the stable version (v1.12.3).
https://github.com/PX4/PX4-Autopilot/issues/19482#issue-1203236704

So using the master here might not be the right solution if this problem appears on all Pixhawk V5X.

Best

Attahayat, 2022-09-28:
> Got the connection to work. For reference, following are the steps:
>
> 1. Connect Ethernet port of Pixhawk 5x to ethernet port of companion computer.
> 2. Set the IP address to static IP for the wired [...]

I do not see the option of Pixhawk Payload Bus. I tried to manually set it in QGroundControl by setting the value for RTPS_CONFIG to 401 (from parameter reference user guide), my pixhawk 5 stops responding and QGroundControl says "disconnected" even when I connect it to my work station via a USB. I had to reset the pixhawk to make it start working again. Any help would be appreciated.