PX4/PX4-Autopilot #19373: Add ability to use 'Tab' to list / choose commands in NuttX shell

junwoo091400, 2022-03-22:
**Describe problem solved by the proposed feature**
Currently, it's quite hard to use NuttX shell (MAVLink Console in QGC) as user has to manually type all the letters in the console correctly. It would be nice to have the commands auto-complete with a 'Tab' input (like in many modern terminals)

**Describe your preferred solution**
Enable Tab commands to be recognized / sent to the Autopilot from QGC (if NuttX supports it).

**Describe possible alternatives**
Having a 'commands metadata' file that shows available commands, which gets transferred from Autopilot (stored in FLASH, like Parameter Metadata) to the QGC, so that QGC can handle the auto-complete GUI, wouldn't be a nice solution. Since [already QGC suffers](https://github.com/mavlink/qgroundcontrol/issues/10204) from outdated metadata issue, and I think moving 'metadata' around should be minimized.

**Additional context**
This will be very useful for PX4 Developers around the world.

If pinpointed in the right direction, I would like to make a PR for this.

dagar, 2022-03-22:
This is already enabled in NuttX on newer boards, although maybe you probably need to use a real serial console at the moment.
[image]

junwoo091400, 2022-03-24:
Any quick idea on how to support this in QGC as well? I would like to add that support if it's manageable :0 @dagar

junwoo091400, 2022-11-09:
Referencing https://github.com/mavlink/qgroundcontrol/pull/9266 to get this implemented.

Ghvhgvhhbgffghg, 2024-04-09:
Hi, please answer my question. 

PX4 have nsh (NuttX shell) in QGround. How type message mavlink use NuttX shell? Thanks.