PX4/PX4-Autopilot #19379: No flight data is logged in SD card sometimes.

Abner0907, 2022-03-23:
Hi, My branch is based on [v1.12.3](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.3).

I have faced a issues that sometimes it will not logged the flight data in the SD card sometimes.  
I am sure it will log the data most of the time.  I am very appreciate that someone has the similar problem and let me know how to fix it.

By the way, I have review the issue  https://github.com/PX4/PX4-Autopilot/issues/18672 but not sure is the same as me.

Abner0907, 2022-03-23:
This is my sd_bench -r 50 test results:
[image]

Abner0907, 2022-03-23:
BTW:  
My hardware is use a private board which is based on a  stm32H743VIT6 chip. 
My card is the below:
[image]

dagar, 2022-03-23:
How reproducible is the problem? Can you try with master (soon to be v1.13)?

spiderkeys, 2022-03-23:
Given that you are on an H743-based board, you might want to check out this PR that was merged back in February. If you can't test on master yet, you should be able to try cherry-picking this commit or manually making the change. One of our boards uses the H743VIT6 as well (MatekH743-Wing), and this changed fixed intermittent SD card issues for us.

https://github.com/PX4/PX4-Autopilot/pull/19177

Abner0907, 2022-03-24:
@dagar  it happens this issues many times, but I have not find the happen regularly. sorry, I can not test v1.13 as soon as possible as some changes in my code.
@spiderkeys  thank you. I will cherry pick your #19177 and test it.

Abner0907, 2022-03-26:
@spiderkeys Hi, I have a manual making the changes.
I found that #19177 can not fix my issues.

@dagar 
I plug the sd card and found the log exist. But the file name is wrong update. This is the raw log. 
https://drive.google.com/file/d/1r5OoeqqW3wnY0qAZCkJxXaU2ZeknpFgu/view?usp=sharing

[image]

Abner0907, 2022-03-28:
@dagar 
another log issues is: while download the log, it will show the error below:
[image]

It seems that something wrong is happened while writing the log. and the log record is stopped. but when the vehicle land , and then arm the drone again(without power off), the log will continue record in the next log file.

closed 2022-04-01 by Abner0907