PX4/PX4-Autopilot #19380: Accelerometer and Gyroscope sensor status

GMahmoud, 2022-03-23:
Hi everyone,

I am working on some integration of PX4 autopilot and I have a question about the state of the following sensors

[MAV_SYS_STATUS_SENSOR_3D_ACCEL 1](https://mavlink.io/en/messages/common.html#MAV_SYS_STATUS_SENSOR_3D_ACCEL)
[MAV_SYS_STATUS_SENSOR_3D_GYRO 1](https://mavlink.io/en/messages/common.html#MAV_SYS_STATUS_SENSOR_3D_GYRO)
When I received the mavlink message SYS_STATUS, these two sensors are neither present and nor enabled however the health state is Good.
Here is the value I am receiving
```sh
[id: 1]
onboard_control_sensors_present:  false
onboard_control_sensors_enabled: false
onboard_control_sensors_health: true
------------
[id: 2]
onboard_control_sensors_present: false
onboard_control_sensors_enabled: false
onboard_control_sensors_health: true
```
I looked a bit in the source code (tag: v1.12.3) and I found that `subsystem_info_s::SUBSYSTEM_TYPE_ACC` and `subsystem_info_s::SUBSYSTEM_TYPE_GYRO`
are used for updating only the health key by using the `set_health_flags_healthy` [method](https://github.com/PX4/PX4-Autopilot/blob/v1.12.3/src/modules/commander/Arming/HealthFlags/HealthFlags.cpp#L77)

Why the update of enable and present is not done as well ?
Is that an issue or I am misunderstanding the SYS_STATUS mavlink message ?

Thanks for your answer
Cheers 😄

dagar, 2022-03-23:
To be honest the mapping of these things to the existing mavlink SYS_STATUS flags was a bit of an afterthought and probably needs a review.

If you actually want to see what PX4 cares about internally start by running `sensors status` on the console.

GMahmoud, 2022-03-29:
Thanks @dagar, I will look at this command related code to do a better integration of these sensors.

closed 2022-03-29 by GMahmoud