PX4/PX4-Autopilot #19382: random RC value causing kill switch without rc receptor attached

NicolasM0, 2022-03-23:
**Describe the bug**
During a flight in offboard mode without RC connected (no RC receptor connected to the autopilot), our drone crashed with "[commander] Kill-switch engaged" message.

Looking at the log, I can see that after 10min of flight some messages are published on input_rc topics and at a moment, that triggers a kill switch.
[image]

The `input_source` is 4: `RC_INPUT_SOURCE_PX4IO_SBUS`. messages are in the table below (removing RC lost messages)

|timestamp|f_timestamp_last_signal|f_rssi|f_rc_lost_frame_count|f_rc_total_frame_count|f_rc_ppm_frame_length|f_values_0_|f_values_1_|f_values_2_|f_values_3_|f_values_4_|f_values_5_|f_values_6_|f_values_7_|f_values_8_|f_values_9_|f_values_10_|f_values_11_|f_values_12_|f_values_13_|f_values_14_|f_values_15_|f_values_16_|f_values_17_|f_channel_count|f_rc_failsafe|f_rc_lost|f_input_source|
|---------|-----------------------|------|---------------------|----------------------|---------------------|-----------|-----------|-----------|-----------|-----------|-----------|-----------|-----------|-----------|-----------|------------|------------|------------|------------|------------|------------|------------|------------|---------------|-------------|---------|--------------|
|726876831|726876831              |38    |0                    |0                     |0                    |1033       |874        |874        |953        |874        |874        |874        |874        |874        |2134       |1356        |874         |1514        |875         |874         |874         |998         |998         |18             |0            |0        |4             |
|726898330|726898330              |38    |0                    |0                     |0                    |1033       |874        |874        |953        |874        |874        |874        |874        |874        |2134       |1356        |874         |1514        |875         |874         |874         |998         |998         |18             |0            |0        |4             |
|737678520|737678520              |38    |0                    |0                     |0                    |874        |2114       |876        |874        |874        |1514       |913        |2114       |874        |890        |874         |952         |2014        |875         |874         |874         |998         |998         |18             |0            |0        |4             |
|760675275|737678520              |38    |0                    |0                     |0                    |1654       |1713       |1513       |874        |874        |874        |874        |874        |1033       |874        |874         |953         |874         |874         |1794        |1558        |998         |1998        |18             |1            |0        |4             |
|768666922|737678520              |38    |0                    |0                     |0                    |1832       |893        |1194       |953        |874        |874        |874        |874        |874        |874        |874         |874         |883         |874         |874         |2153        |998         |998         |18             |1            |0        |4             |
|774278867|774278867              |38    |0                    |0                     |0                    |874        |874        |876        |874        |874        |874        |1994       |2153       |2074       |893        |874         |1913        |883         |874         |1994        |878         |998         |998         |18             |0            |0        |4             |
|787076723|787076723              |38    |0                    |0                     |0                    |1032       |874        |1512       |1514       |1523       |875        |2014       |878        |874        |874        |1354        |949         |874         |874         |912         |2074        |998         |998         |18             |0            |0        |4             |
|787469746|787469746              |38    |0                    |0                     |0                    |994        |874        |1494       |874        |879        |874        |874        |874        |874        |874        |874         |874         |2114        |875         |1994        |1513        |998         |998         |18             |0            |0        |4             |

When the first input rc message is published we have a pwm_out message: `[pwm_out] instance: 0, max rate: 100, default: 50, alt: 50`

**Log Files and Screenshots**
I am not able to upload the log flight review (error: `500 Internal Server Error`). The log is here : https://drive.google.com/file/d/1fnVp8m0kPGOqecR0gfb8Nm3PBm5HBsQN/view?usp=sharing

**Additional context**
PX4 version: 1.12.3 + very few changes
hardware: CUAV V5+

This crash happened yesterday and we reproduced the same issue on the ground today on the same board.

We really need to understand what happened because we have some systems flying with the same configuration.
Any help would be very appreciated.

dagar, 2022-03-23:
I don't know what's going on electrically, but you're periodically getting supposedly valid SBUS data.

[image]

For actual switch handling there are simple protections in place to try and catch things like RC input source changes, and then requiring back to back valid switch interpretation before publishing. I'll see if I can tighten that up a bit.

https://github.com/PX4/PX4-Autopilot/blob/e6ed59565184d8bbf2da316f6c1cbe4b14fb366e/src/modules/rc_update/rc_update.cpp#L647-L662

Then within the px4io we can add some basic protections like not publishing RC_INPUT_SOURCE_UNKNOWN.

At one point I was considering introducing a median filter per RC channel, but there were latency concerns.

dagar, 2022-03-23:
Possible fixes in https://github.com/PX4/PX4-Autopilot/pull/19385. If there's any underlying root cause in the SBUS decoding, px4io, etc I'm still interested in finding and addressing it properly, but that we can also improve our tolerance for these things at all levels.

NicolasM0, 2022-04-11:
Hi @dagar,
Thank you for all your recent changes that will make this issue very unlikely to reproduce.
On my side, I have made sure that RC is disabled on our vehicles without RC on the autopilot (on px4 v1.12).

I close this issue but I have no idea why it happened on only one of our autopilots.
I would like to estimate the probability of this issue on vehicles that we could not update for now... Do you have any idea about this?

If you want to investigate more, I can run some tests on my defective hardware.

dagar, 2022-04-11:
> I would like to estimate the probability of this issue on vehicles that we could not update for now... Do you have any idea about this?

I'm not sure, but @davids5 might be able to comment on what's potentially happening electrically. 

We're still pursuing more fixes like in the NuttX serial driver parity checking (https://github.com/PX4/PX4-Autopilot/pull/19446), but ultimately I want to move away from all input scanning entirely. I'd like to make the RC protocol (and port, etc) an explicit configuration. Then in cases where you have nothing physically connected you can simply adjust the parameter and none of that code will even execute.

It's a bit of a hack, but if you have existing systems that can't be upgraded immediately you could either stop the `rc_update` module if you want no possibility of manual control getting through or clear the RC_MAP* parameters. Connecting something to the RC input might also be a possibility.

davids5, 2022-04-11:
Placing a 10K Resister from the input to ground may also work. It would be work a try to see if it is really related to the HW.

NicolasM0, 2022-04-11:
> Placing a 10K Resister from the input to ground may also work. It would be work a try to see if it is really related to the HW.

I will do that

closed 2022-04-20 by NicolasM0