PX4/PX4-Autopilot #19390: Simulator for Fixed wing

airsimdevdd, 2022-03-24:
Is there any simulator in which we can simulate fixed-wing and import our model dynamics to it custom model dynamics to it ,also multi-vehicle support ,it can be open source or paid

Jaeyoung-Lim, 2022-03-24:
@airsimdevdd I think the default PX4 Gazebo Software In the loop simulation framework supports all the things you have mentioned. Is there something missing?

airsimdevdd, 2022-03-24:
I'm unable to do it in gazebo , I tried generating the sdf from blender using a addon called phobos  even -though  
everything is checked the  generated sdf makes some errors like exploding parts(solved now)and vehicle not spawning on the ground it is spawning on  some  blank surface without any axis and one more reason graphics quality is compromised on gazebo . could anyone please help me out that would be great ,if you can't suggest me a good simulator which is better and easy import my custom model dynamics

airsimdevdd, 2022-03-24:
There are many of my colleagues too facing issues in gazebo sdf for import their custom vehicles to gazebo

Jaeyoung-Lim, 2022-03-24:
@airsimdevdd Could you maybe explain what exactly they are struggling with? If you still can't resolve the issues you can reach me via the PX4 slack: https://slack.px4.io/

All the supported simulators are in https://docs.px4.io/master/en/simulation/, However I would say that gazebo is the easiest to experiment and play around with custom dynamics.

closed 2022-03-27 by Jaeyoung-Lim