PX4/PX4-Autopilot #19392: PX4 Aux output ports not outputting PWM signal when using RC passthrough 3/24/22

tyrian869, 2022-03-24:
**Describe the bug**
I've mapped under the radio tab on herelink AUX2 to channel 7 that is clearly moving under the channel monitor

**Expected behavior**
To allow me to control Retract controller that is expecting the PWM signal.

**Drone (please complete the following information):**
Generic 10" Octo coaxial geometry, because there was nothing bigger then a 10"................. 

**Additional context**
I've read that PX4 airframe reference was important because if it did not show aux ports with rc passthrough this would not work? not sure why that matters as we have the option of configuring these ourselves. 

I've read that people who are doing PWM camera triggers say that they can get the camera to shutter when using trigger PWM signal  on all aux ports 1-6 which is true for me as well.  

I read also that the system has to be armed for both camera shutter trigger test and RC passthrough to work, this was not true for camera shutter and tried it anyway for RC passthrough but that did not work.

I've tried also setting up landing gear under flight mode as well, as well as  just assign the RC channel to the aux port directly. nether worked. my tests are on the ground.

I might connect a scope up to the AUX port to see if it's doing anything, I've connected a servo but it does nothing. 
I do have +5v connected to the output rail as well incase anyone is wondering. 

Can't believe I'm here again.

Hope this helps you help me help others.

spiderkeys, 2022-03-24:
I don't know if it is exactly the problem you are having, but there are some similar elements to issues I was having with trying to control servos via RC Passthrough under the new control allocator system. I'd be interested to learn what the current 'proper' way of achieving control through RC Passthrough of aux servos is currently supposed to look like... Are you using Control Allocator on your airframe?

Screenshots of my problem description from a slack thread, if it ends up being related:
[image]

[image]

[image]

tyrian869, 2022-03-25:
It's not that for this instance we couldn't use Mavlink to make this work,  I was more or less just trying to use what is already in place, this is for a landing gear controller that takes in the PWM signal, to start powering dc brushed motors that are worm driven, that uses amp out technology to stop, old school but for a test bench it's just fine. this actuator outputs menu, I have not seen this before, is that part of QGC? (Q ground control)

spiderkeys, 2022-03-25:
Yes, the actuators tab will appear in QGC if you are running PX4 with the new control allocation system enabled (SYS_CTRL_ALLOC=1). If you don't see the tab, then the behaviors I saw are probably unrelated. 

Looking at the your airframe's startup scripts, it looks like it still defaults to the old mixer system, so this probably doesn't apply.

closed 2022-10-25 by bkueng