PX4/PX4-Autopilot #19394: Can not keep offboard mode!

1556260456, 2022-03-25:
I have used linux (ubuntu) to send location information to PX4,and I change position mode to offboard mode by RC，The interface used is based on MAvlink and serial port[https://github.com/mavlink/c_uart_interface_example].
Before this, it worked normally. Now, after running the program that sends instructions on the Linux terminal, I switch to the offboard mode on the remote control, and it is almost impossible to enter the offboard state, Once in, it will quickly switch out.
“Reject offboard” displays in QGC. Occasionally, “failsafe Action, failsafe deaction,no offboard mode” may be displayed. How should I check and modify to enter offboard mode normally?
Thanks!

1556260456, 2022-03-26:
I have fixed this problem。But I have met a new one,I send velocity information to PX4 at 50hz, before it's 4hz。 is it too fast？

omerts, 2022-04-11:
@1556260456 Can you please share your fix?