PX4/PX4-Autopilot #19396: No such file or directory 'Kconfig'

dookei, 2022-03-25:
Hi I am trying to build on nvidia jetson machine the current master. I am getting this error.
```
-- PX4 version: v1.13.0-alpha1-4498-g5ae4cae073
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3") 
-- PX4 config file: /home/jetson/catkin_ws/src/PX4-Autopilot/boards/holybro/durandal-v2/default.px4board
/home/jetson/.local/lib/python3.8/site-packages/defconfig.py: [Errno 2] No such file or directory: 'Kconfig'
/home/jetson/.local/lib/python3.8/site-packages/genconfig.py: [Errno 2] No such file or directory: 'Kconfig'
CMake Error at cmake/kconfig.cmake:58 (file):
  file STRINGS file
  "/home/jetson/PX4-Autopilot/build/holybro_durandal-v1_default/boardconfig"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

CMake Error at cmake/kconfig.cmake:360 (include):
  include could not find load file:

    px4_impl_os
Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

CMake Error at cmake/kconfig.cmake:361 (px4_os_prebuild_targets):
  Unknown CMake command "px4_os_prebuild_targets".
Call Stack (most recent call first):
  CMakeLists.txt:174 (include)

-- Configuring incomplete, errors occurred!
Error: /home/jetson/PX4-Autopilot/build/holybro_durandal-v1_default is not a directory
make: *** [Makefile:230: holybro_durandal-v2_default] Error 1
```
I installed the kconfig lib over pip with 
`pip3 install kconfiglib`
Still I dunno why it cant find kconfig
Any thoughts what I might be missing?

closed 2022-03-25 by dookei

teddy590327, 2022-05-16:
I have the same problem, did you figure out what was the problem? Or anyone has any idea why this happen and how to solve it?

rizvanovic, 2022-05-27:
Experiencing this too. Haven't been able to fix it.

dagar, 2022-05-27:
@rizvanovic @teddy590327 did you install kconfiglib? https://github.com/PX4/PX4-Autopilot/blob/1ba3610cfc7aeaf8ab12e65e53f0d60ef391f639/Tools/setup/requirements.txt#L9

Try running the setup scripts (https://github.com/PX4/PX4-Autopilot/tree/master/Tools/setup). eg for ubuntu `./Tools/setup/ubuntu.sh`.

prelude404, 2022-10-22:
did you fix this problem? I am experiencing the same problem now