PX4/PX4-Autopilot #19404: Fixed-wing airframe cannot takeoff in Gazebo HITL

huhuhuq, 2022-03-26:
when running HITL using the HILStar plane, the plane cannot take off, even with a mission loaded, the plane armed and the plane in "Mission" mode. But SITL work well.

huhuhuq, 2022-03-26:
Can gazebo HITL support HILStar plane airframe?

jarelias, 2022-05-03:
Hi @huhuhuq , I hope you get an answer since I am in a similar situation. As far as I can see there is no gazebo model already built to support HITL but you can take the standard plane and modify some things in the sdf file to make it HITL compatible, for that you can take the hil_standard_vtol to see what you need to change. That worked for me however it only works with the stable firmware version, so if I want to make some modification on the HilStar or if I want to add a new ariframe, then HITL doesn't work. I also tried running hil_standard_vtol that is already built in by default and works well with stable firmware in the pixhawk but if a make a new firmware and upload it then HITL doesn't work even if I don't make any modifications.

This is a link to my post regarding the same issue:
https://discuss.px4.io/t/fixed-wing-hitl-with-gazebo/27206?u=jesus

Maybe you can post one too and join to slack to push this forward because I have seen posts that remain there for years and never get answered.

closed 2025-10-29 by github-actions[bot]