PX4/PX4-Autopilot #19405: USB device not detected after boot(current Master branch)

dookei, 2022-03-28:
Hi there,
I am currently facing an USB issue on the master branch with the Holybro Durandal_v1 hardware.
After boot the USB is no longer detected on the host machine. During boot(or when the STM32 enters the bootloader) this is detected. So I can rule out hardware or cable issues. This is only reproducible on the current master branch.

On the previous firmware v1.12.3 this does not happen.
Any thoughts where I should look for ?
Many thanks

davids5, 2022-03-28:
@dookei - I just verified this as working on PX4 master with today's Linux QGC build. 

```
ercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
/dev/ttyACM0: launching mavlink (HEARTBEAT v2 from SYSID:255 COMPID:190)
INFO  [mavlink] mode: Config, data rate: 100000 B/s on /dev/ttyACM0 @ 2000000B
[boot] Rev 0x0 : Ver 0x0 VD00
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: HOLYBRO_DURANDAL_V1
HW type: VD00
HW version: 0x00000000
HW revi[hardfault_log] Fault Log is Armed
sion: 0x00000000
FW git-hash: 67920f089bdb73e0609338a9c91ad4071a3f21d9
FW version: 1.13.0 40 (17629248)
FW git-branch: master
OS: NuttX
OS version: 0.0.0 0 (0)
OS git-hash: c5c7d2b4f26f52f1dfb425ebde83328606b65d4f
Build datetime: Mar 28 2022 10:00:56
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00060000000033383331353751100035002c
MCU: STM32H7[4|5]xxx, rev. Y
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [param] importing from '/fs/mtd_params'
WARN  [parameters] ignoring unrecognised parameter 'CAL_ACC0_TEMP'
WARN  [parameters] ignoring unrecognised parameter 'CAL_ACC1_TEMP'
WARN  [parameters] ignoring unrecognised parameter 'CAL_GYRO0_TEMP'
WARN  [parameters] ignoring unrecognised parameter 'RC_ACRO_TH'
WARN  [parameters] ignoring unrecognised parameter 'RC_ASSIST_TH'
WARN  [parameters] ignoring unrecognised parameter 'RC_AUTO_TH'
WARN  [parameters] ignoring unrecognised parameter 'RC_MAN_TH'
WARN  [parameters] ignoring unrecognised parameter 'RC_STAB_TH'
INFO  [parameters] BSON document size 1402 bytes, decoded 1402 bytes (INT32:24, FLOAT:44)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 62560 bytes
Loading airframe: /etc/init.d/airframes/4014_s500
INFO  [uavcan] Node ID 3, bitrate 1000000
ERROR [uavcan] CentralizedServer init: -13
ERROR [uavcan] UavcanServers init: 0
INFO  [px4io] IO FW CRC match
Board sensors: /etc/init.d/rc.board_sensors
icm20689 #0 on SPI bus 1 rotation 2
bmi088_accel #0 on SPI bus 1 rotation 2
bmi088_gyro #0 on SPI bus 1 rotation 2
WARN  [SPI_I2C] Already running on bus 1
ERROR [SPI_I2C] bmi088: no instance started (no device on bus?)
WARN  [SPI_I2C] Already running on bus 1
ERROR [SPI_I2C] bmi088: no instance started (no device on bus?)
ist8310 #0 on I2C bus 3 address 0xE rotation 10
ms5611 #0 on SPI bus 4
WARN  [SPI_I2C] ist8310: no instance started (no device on bus?)
ERROR [heater] Valid SENS_TEMP_ID required
INFO  [pwm_out] instance: 0, max rate: 100, default: 50, alt: 50
INFO  [init] Mixer: /etc/mixers/quad_x.main.mix on /dev/pwm_output0
INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1
ekf2 [690:237]
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS1
WARN  [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
INFO  [logger] logger started (mode=all)

NuttShell (NSH) NuttX-10.2.0
nsh> /dev/ttyACM0: launching mavlink (HEARTBEAT v2 from SYSID:255 COMPID:190)
INFO  [mavlink] mode: Config, data rate: 100000 B/s on /dev/ttyACM0 @ 2000000B
INFO  [mavlink] Starting mavlink shell
```

Perhaps you can give a  more complete description of your setup.

RamanaBotta, 2022-03-29:
same problem.

dookei, 2022-03-29:
Hi @davids5 I wont be able to check it today as I have no access to the FC at the moment. I will get back to you tomorrow.

vincentpoont2, 2022-03-30:
Hello @RamanaBotta & @dookei ,  Can you go to Mavlink Console and type in `ver all` and paste the print out here?
[image]

dookei, 2022-03-30:
Hi @vincentpoont2 
Here is my output 
```
NuttShell (NSH) NuttX-10.2.0
nsh> 
nsh> ver all
HW arch: HOLYBRO_DURANDAL_V1
HW type: VD00
HW version: 0x00000000
HW revision: 0x00000000
FW git-hash: df0e402c447c47458a29472c493cbfa035ae68cf
FW version: 1.13.0 0 (17629184)
FW git-branch: master
OS: NuttX
OS version: Release 8.2.0 (134349055)
OS git-hash: 017aa15746bcf6b57bb185e14dce7fa0876fc751
Build datetime: Mar 30 2022 09:12:51
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000343434363038510300450027
MCU: STM32H7[4|5]xxx, rev. Y
nsh> 
```
I have this FC connected to a nvidia jetson all the time over USB and UART . In the previous version v1.12.3 I could connect to a mavlink term over the USB port.
This is my lsusb output for the Durandal during boot (when the STM is in bootloader)
Bus 001 Device 012: ID 3162:004b
After the bootloader when it enters the Px4 the USB device disappears.

dookei, 2022-03-30:
I believe this is the issue ...
[image]
on this commit the USB was disabled:https://github.com/PX4/PX4-Autopilot/commit/6d78054f5001548ea2f1d48bb59966e2352bc9fe

@dagar would it be possible to enable this back?

davids5, 2022-03-30:
@dookei - the usbvbus detection is will automatically start the mavlink over USB once VBUS is active AND the far end opens the port and sends mavlink.

Do you have a serial debug console?

Please run on the console
```
listener system_power

TOPIC: system_power
 system_power
    timestamp: 32652442 (0.003585 seconds ago)
    voltage5v_v: 4.9773
    sensors3v3: [3.3065, 0.0000, 0.0000, 0.0000]
    sensors3v3_valid: 1
    usb_connected: 1
    brick_valid: 0
    usb_valid: 1
    servo_valid: 1
    periph_5v_oc: 0
    hipower_5v_oc: 0
    comp_5v_valid: 0
    can1_gps1_5v_valid: 0
```
and post the results here.

Can you also measure the voltage level of the USB +5V?

dookei, 2022-03-31:
Hi @davids5 , that's easy. I have no USB power. 
I have my FC connected to a USB hub without power line. Just data and ground. Reason for it is to guarantee that the FC(and peripherals) are not powered over the USB.
On the jetson side I just use then usbguard to block or allow the usb port communication.

Here is the output 
```
nsh> listener system_power
TOPIC: system_power
 system_power
    timestamp: 602587539 (0.010019 seconds ago)
    voltage5v_v: 5.0427
    sensors3v3: [3.3090, 0.0000, 0.0000, 0.0000]
    sensors3v3_valid: 1
    usb_connected: 0
    brick_valid: 3
    usb_valid: 0
    servo_valid: 1
    periph_5v_oc: 0
    hipower_5v_oc: 0
    comp_5v_valid: 0
    can1_gps1_5v_valid: 0
```
Would't it be possible to allow the sercon /usb communication when USB communication is available? Instead of just checking if the USB is powered. Its probably more tricky but would be more reliable.
There can also be cases where someone (not recommended)just uses a Power USB cable to power the FC and not to communicate with it.

davids5, 2022-03-31:
The FC acting as a USB device and is looking for VBUS to begin the enumeration. IIRC there would need to be changes in the nuttx driver and PX4 to support this. change. A quick fix would be to add a 100K-1M resister to pull the USB +5. It will pass the sense test and not power the FC.

dookei, 2022-04-01:
ok I will keep in mind next time I have physical access to the FC to add a pull-up to the VBUS power line.
For the time being I have activated the sercon and mavlink on my local branch. I have no other way to do it at the moment.

Perhaps you can give me you opinion on this.
I am currently developing a carrier board( privately, not going corporate on this one :wink:)  with two FCs(STM32H7) and a nvidia jetson(this could also operate as FC). 
My idea was, besides having all three in a local 100M network, to have the FCs connected over USB to the jetson. So that in case something goes wrong with firmware upload over ethernet can still have direct access to the usb bootloader from the stm32.
For that I can't(well I can just I am trying to avoid) have the FCs VBUS being powered (similar situation to now with the durandal).
I am using a USB passive switch to jump between usb devices connections.  Just avoiding using usb hub, since the FCs connections are only needed for firmware upload.
This is what I am using: 
[image]
Where USB2_CH2 and USB2_CH3 are connected to the FC/STM32s. The STM32 are constantly powered. 
**My dilema:** 
I am currently wondering if this is good practice. Since I have the FCs constantly powered on and they dont have VBUS power line. What you think ? Bad? good?  :confused:

closed 2022-04-06 by dookei

dookei, 2022-04-06:
Thanks I will try it next time I have phisical access to the FC. 
Many thanks @davids5 !