PX4/PX4-Autopilot #19427: Control allocator issue: how to set to disarmed state or reject old topics

dookei, 2022-03-31:
Hi there, 
so I am using the control allocator to control 6 servos plus two power rails mosfets(they just power the Servos).
The idea with the power rail mosfets is to power the servos on demand.  
**The Problem**
Once I published to the servos and this have values stored(even if they are old) and later on I power the servos, this are still be active to the current stored topic value.

**Example:**
**1.** Power rail on
**2.** Servo_1 set to 1(or 2000)
**3.** Power rail off

Note:At this point the topic from servo_1 still contains the value 1.
Now I want to activate Servo_2 but not Servo_1.

**1.** Power rail on
**2.** Servo_2 set to 1(or 2000) ---> the Servo_1 is going to be active since this is still receiving PWM.
**3.** Power rail off
**Questions**
**1.** Is there a way that the mixer_module or control allocator just generates output if the value published to the affected topic is recent(lets say 1ms)? 
OR
**2.** Is it possible to publish a value that sets the servos to the disarmed value(in my case 0 zero)? PWM_MAIN_DISx or PWM_AUX_DISx

@bkueng: I believe you could be of great help here once again ;-)

PS: a work around that I did not yet test, would be to set the paramenters PWM_MAIN_MINx /PWM_AUX_MINx to zero and find the working range from the servos. This way I could set the topic to -1(zero PWM) when needed.  But this would be a sloopy/hacky way that can lead to missuses.

bkueng, 2022-04-01:
> 1. Is there a way that the mixer_module or control allocator just generates output if the value published to the affected topic is recent(lets say 1ms)?

No, there's no generic value that would work everywhere. You could do it in your specific subscription. But better not to have too complex logic in there, and let the publisher update to the value you want.

> 2. Is it possible to publish a value that sets the servos to the disarmed value(in my case 0 zero)? PWM_MAIN_DISx or PWM_AUX_DISx

Yes if you use NaN.

dookei, 2022-04-01:
Many thanks for your fast reply!
Your rigth. I will go with solution #2. 
do mean this NAN ?
```#define NAN         (0.0/0.0)```

bkueng, 2022-04-01:
> do mean this NAN ?

Yes

dookei, 2022-04-01:
Ok thanks. I will give it a try. Have a nice weekend!

closed 2022-04-01 by dookei