PX4/PX4-Autopilot #19432: KakuteH7 QGC PX4 Master Installation Failed

vincentpoont2, 2022-04-01:
**Describe the bug**
With the bootloader `make holybro_kakuteh7_bootloader` flashed onto the board via betaflight configurator. Unable to install PX4 Master via QGC.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash bootloader `make holybro_kakuteh7_bootloader` to board with betaflight configurator
2. Connect to QGroundcontrol -> advanced settings -> install master

**Expected behavior**
Install KakuteH7 Master Firmware

**Log Files and Screenshots**
[image]

bkueng, 2022-04-04:
The board needs to be added to QGC, like this: https://github.com/mavlink/qgroundcontrol/pull/8058
(we really need to make this more generic)

vincentpoont2, 2022-04-06:
@mrpollo @bkueng Would you be able to assist to on helping the board added to QGC? 

I don't understand how it works in the back end. I tried http://px4-travis.s3.amazonaws.com/Firmware/stable/holybro_kakuteh7_default.px4 and it doesn't work. Does the firmware need to be uploaded first?

mrpollo, 2022-04-07:
Hey @vincentpoont2 CI automatically builds and uploads to S3 for us, it seems we forgot to include the new board to the list of boards, I fixed that on #19453

closed 2022-04-07 by dagar

vincentpoont2, 2022-04-08:
@mrpollo is this https://github.com/mavlink/qgroundcontrol/pull/8058 still the correct way to link a board ID to the firmware? 
A lot of the boards aren't on here   [  // Supported bootloader board ids](https://github.com/mavlink/qgroundcontrol/blob/c7c070cfb1fd8fdf4f855f1c5c56d731a1081222/src/VehicleSetup/Bootloader.h#L63) . Would you be able to help us link the Board ID to the firmware S3 link?

[image]

Sam2636, 2023-09-27:
[image]
I am getting no gyro found for kakuteh7  how to solve this issue?

vincentpoont2, 2023-09-27:
> [image]
> I am getting no gyro found for kakuteh7  how to solve this issue?
> 

Which version firmware did u install and which version of kakute h7 do you have?

julianoes, 2023-10-18:
Could this be related to https://github.com/PX4/PX4-Autopilot/issues/22236?

PR #19453 (2022-04-07): ci: build and deploy kakuteh7
Adds holybro kakuteh7 to Travis CI for building and deployment to S3 for QGC

fixes #19432 

cc @vincentpoont2