PX4/PX4-Autopilot #19433: Arming but not taking-off in SITL

MrPrietix, 2022-04-01:
I am setting the drone in Offboard mode and arming it when it is in a moving platform. The drone arms and takes-off if it is in a steady platform. However, when the platform moves at 2-3m/s then the drone is able to spin the propellers (armed) but it does not take off. Seems like a bug.

The px4 version that I am using is the 13.0.

Jaeyoung-Lim, 2022-04-01:
@MrPrietix Can you provide a log?

MrPrietix, 2022-04-04:
> @MrPrietix Can you provide a log?

MrPrietix, 2022-04-04:
Hi @Jaeyoung-Lim, yes, sure ([log file](https://logs.px4.io/plot_app?log=853f586b-773b-46a5-994e-7f2d5c7e4338)). Thanks for the help.

Jaeyoung-Lim, 2022-04-04:
@MrPrietix Which setpoints are you sending in offboard mode? From the logs it seems as if you are not sending any valid offboard setpoints

MrPrietix, 2022-04-04:
First of all, I am using the position controller in off-board mode with a feed-forward velocity term by setting velocity and the position field to true in the offboard_control_mode UORB message. Consequently, I send the desired velocity and a position using the topic trajectory_setpoint which is based on the UORB message vehicle_local_position_setpoint. The velocity that I send in offboard mode is 2m/s in the y axis of the drone to follow the platform while taking off. Regarding the position, I am sending the x and y position of the platform while setting z at - 10m to take-off. Furthermore, in every velocity/position sent, I am also sending the timestamp.

Let me know if you need anything else.

Jaeyoung-Lim, 2022-04-04:
@MrPrietix The log shows that the setpoints doesn't seem to be arriving to the position controller (the setpoints are invalid). 

Can you verify that you can fly in offboard mode with your setup without the moving platform?

MrPrietix, 2022-04-04:
[This log file](https://logs.px4.io/plot_app?log=6a5d16e2-6b3e-4fca-8df7-172d3d1a5108) is the one with the static platform.
[This log file ](https://logs.px4.io/plot_app?log=6b5e44ba-9fc1-4464-9f5e-572b4132fc16)is with the moving platform but at 1m/s.

Jaeyoung-Lim, 2022-04-04:
@MrPrietix On the logs it shows that the vehicle was able to take off on both static and moving platforms, but for some reason the vehicle switches back to mission mode after the takeoff.

As far as I can see, for the original log, there is no position setpoints logged and therefore the vehicle is not taking off. If you are saying that the rest of the setup is identical but ONLY the moving platform is different between the two experiments, I would try to find out why your setpoints are either being invalidated or not being used by the position controllers

MrPrietix, 2022-04-04:
@Jaeyoung-Lim Once it has taken off and reached some altitude I set it in mission mode to perform a mission. Do you have any clue where could I start the debugging process, such as cpp files. Maybe it is due to some parameters in PX4.

closed 2022-04-05 by MrPrietix

Jaeyoung-Lim, 2022-04-05:
@MrPrietix Is the issue fixed? Could you share on how you have resolved the issue?

MrPrietix, 2022-04-08:
Hi @Jaeyoung-Lim , yes it is fixed, just needed to change the parameter LNDMC_XY_VEL_MAX to the velocity of the platform

omerts, 2022-04-11:
I also ran into this, and nothing helped but the following param values:
NAV_DLL_ACT = 0
NAV_RCL_ACT = 0 
COM_RCL_EXCEPT = 4

Taken from: https://discuss.px4.io/t/offboard-mode-in-sitl/25727/3