PX4/PX4-Autopilot #19442: Compass calibration should detect sufficiently misaligned orentation

hamishwillee, 2022-04-06:
If the compass is aligned at yaw 30 and you run a calibration that will succeed on PX4 1.12/master. However since yaw 30 is not supported, the calibration will select the nearest supported case - yaw 40. The result is that the calibration appears to succeed but is invalid, leading to warnings etc.

The bug is that this is silent. Assuming the calibration is capable of detecting the orientation it should provide a warning/fail if the error in orientation detection  is too great. 

This comes out of https://github.com/PX4/PX4-user_guide/issues/1833 . I have "fixed" in the documentation, but really either we should support any orientation or we should warn on false orientations.

bresch, 2022-04-06:
> Assuming the calibration is capable of detecting the orientation

The problem is that the detection does not estimate the orientation, it's simply trying all available orientations and selects the one with the smallest output error. So "40" is not selected because it is close to 30 (we don't know that this is the true orientation), but because the mag has the smallest error compared to the internal one when rotated by 40 degrees.

Finwood, 2022-04-06:
> it's simply trying all available orientations

Are there plans to move sensor orientations to free angles eventually? If not, would it be possible to add more supported orientations? We have a hexacopter airframe, which is notoriously heavy on 60° and 30° angles 😅

Finwood, 2022-04-06:
How does the EKF play into this? If I had, say, a 5 degree offset in my compass mount, would PX4 eventually learn this offset and compensate for it, or would all estimations continue being 5° off?

bresch, 2022-04-08:
> Are there plans to move sensor orientations to free angles eventually?

Yes. It won't be tomorrow, but we already discussed about it

> If not, would it be possible to add more supported orientations? We have a hexacopter airframe, which is notoriously heavy on 60° and 30° angles

Yes, you can add your own proffered rotation in the code, add it to the rotation enum and use it for your airframe

> How does the EKF play into this? If I had, say, a 5 degree offset in my compass mount, would PX4 eventually learn this offset and compensate for it, or would all estimations continue being 5° off?

No, the EKF learns the magnetic bias, not the misalignment of the sensor. It will try to learn a bias to correct the misalignment but it won't compensate the error properly. The error might be temporarily reduced when the heading is observable but it's always better if you can set the correct orientation.

closed 2025-10-01 by hamishwillee