PX4/PX4-Autopilot #19443: Matek H743 doesn't autoconnect to QGC

giacomo892, 2022-04-06:
**Describe the bug**
Matek H743 running master doesn't autoconnect to QGC when plugged via USB. Only way is to force a new comm link on the USB and then it connects.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash master on H743-Mini board
2. Open QGC and wait, nothing happens

**Expected behavior**
Board to connect

The board is completely silent con ttyACM0 while checking with screen so I guess that's why QGC doesn't autodetect a board being connected.

taileron, 2022-04-08:
The CDCACM product and vendor entries (DEFCONFIG) are not yet recognized by QGC Daily via USB.
The Flash procedure gets its board ID from somewhere else and is not affected by these entries. 
Currently you can only create a com link e.g. with autoconnect or fake these entries.

giacomo892, 2022-06-22:
This is still an issue now that v1.13 stable has been released

giacomo892, 2022-06-24:
> The product ID of this h743 boards should finally be 0x03f5 (hex value of board id:1013) and the vendor ID could be 0x120A the same as Matek gns-m9n. The next step is that this values come into QGC as well ... temporarily in the h743-mini board code spoofing values are used that QGC knows already

I cannot get the board connected unless I mode QGC to scan for "Matek" name in the USBBoardInfo.json file

z1047941150, 2022-07-16:
I cannot get the board connected QGC until today

lotsofone, 2023-02-14:
I cannot get the board connected QGC until today
h743 wing v3

lotsofone, 2023-02-14:
Successfully connected matek h743 wing to QGC with Siyi fm30 bluetooth

junwoo091400, 2023-08-10:
I have created workaround fix PR here: https://github.com/PX4/PX4-Autopilot/pull/21957

But if we can figure out the proper vendor ID and product ID, then update QGC's Json file, we can do it properly as well.

However, users won't be able to connect to older versions of QGC without that change, and I guess this experimental board users should live with that :shrug: