PX4/PX4-Autopilot #19444: [Project Tracker] Control Allocation User experience improvement

junwoo091400, 2022-04-06:
## Description
This is an issue to track tasks & have discussions on improving the Control Allocation user experience. I previously had a google doc to keep these tracked, but now making it public on Github repo to make it more transparent!

### Why
The Control Allocation is already implemented : #18776 and after the upcoming release of PX4 v1.13.0, it will be the main controller. So making the user experience as smooth as possible is important, so that every user can set it up without confusion.

## Discussions

## TODOs
- [ ] Doc : Add Description on Control Surface
- [ ] Doc : Add Description on setting 1500 as Servo's disarmed PWM output value to avoid weird motions
- [ ] Doc : Add 'Center of Gravity' diagram to show where the distance measurement reference point is (It's not the Pixhawk's center!!!)
- [ ] Doc : Add description on the 'Json' file being transferred from PX4 to QGC, describing actuators setup, to explain what 'dynamic loading' of Actuators configuration actually means
- [ ] Doc : For VTOL / Fixed Wing : Add description for enabling 'lock control surface to hover' and 'commander prearm' settings, to properly check trimmed servo outputs.
- [ ] QGC : Add 'hz' indicator to PWM 50, PWM 100, etc selections in Actuators tab to avoid confusion on what those numbers mean
- [ ] QGC : Put warning 'pop up' when user inputs a value (e.g. Roll torque input for Left Elevon ...) that doesn't make sense according to control surface convention, to avoid user confusions
- [ ] QGC : Add support for basic fixed wing /  rover graphics for Actuators tab (currently only multi-copters are visualized)
- [ ] QGC : Fix bug on initial graphics of multicopter not showing anything when motor positions are (0,0)
- [ ] QGC : Make trim part click-adjustable, e.g. by a unit of 0.05 or so.

Jaeyoung-Lim, 2022-04-06:
@junwoo091400 There is actually a fixedwing model in Gazebo SITL missing for control allocation. Would be great if we can have this as an example as part of this :smile:

junwoo091400, 2022-04-06:
If you could point me to where I could check, that would be great! @Jaeyoung-Lim

Jaeyoung-Lim, 2022-04-06:
@junwoo091400 We can currently test control allocation models in SITL with the following two models: https://github.com/PX4/PX4-Autopilot/tree/master/ROMFS/px4fmu_common/init.d-posix/airframes

- [10017_iris_ctrlalloc](https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/init.d-posix/airframes/10017_iris_ctrlalloc)
- [6012_typhoon_h480_ctrlalloc](https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/init.d-posix/airframes/6012_typhoon_h480_ctrlalloc)

However, we are missing a fixedwing model as an example

bkueng, 2022-04-06:
@Jaeyoung-Lim you can enable it here: https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/init.d-posix/airframes/1030_plane#L46=

junwoo091400, 2022-05-06:
Control Surfaces & Servo Documentation done in https://github.com/PX4/PX4-user_guide/pull/1858/commits/a81088a3c65bfa18632c7fcf8bbe6045f9070f55

junwoo091400, 2022-06-10:
Added TODOs on Control Surface explanation, as the [user ](https://px4.slack.com/archives/C581ST17V/p1654864120332779?thread_ts=1654863386.548499&cid=C581ST17V)was having trouble setting it up

[image]

Tiktiki, 2022-06-26:
Rebuilding a Gen1 Firefly6 and I was looking at the geometry settings.
Are the values (in meters) from the CG or from the IMU?
On this specific airframe both are relatively close, but this is a stepping stone to a fully custom VTOL I am working on.

ALSO

Having the ability to select which servo/gear output gets enable in pre-arm would be great. I have issues with landing gears moving when already set in pre-arm. For example I have a plane with gear that can be catapult-launched or take-off from a runway. Depending on the situation the gear need to be up or down at pre-arm/arming.

junwoo091400, 2022-11-02:
A lot of the QGC improvements would be possible after my Frames metadata QGC PR goes in, as I would have way more Qt knowledge & good idea on how to implement the changes in the TODO list 😉 

https://github.com/mavlink/qgroundcontrol/pull/10472

closed 2025-07-02 by junwoo091400