PX4/PX4-Autopilot #19457: Unable to launch Custom Airframe (proprietary) on SITL.

REGATTE, 2022-04-08:
Hi,

label -> gazebo

I'm trying to build a custom airframe for **PX4 SITL**. I've got all the `airframes. mixer and TOML` files sorted out, 
Made sure all files are added in `CMakeLists.txt` wherever necessary. 

The model is sort of a standard-vtol but with octa-coax and more control surfaces, so made sure to build the `airframe and airframe.post` for SITL w.r.t the *`typhoon`*. I've sorted out all parameter issues, and i get the following error. I've tried looking through others issues but couldn't figure out how to solve this.

Terminal Output below:
```
ashok@ubuntu:~/Desktop/PX4-Autopilot$ VERBOSE_SIM=1 HEADLESS=1 make px4_sitl gazebo_mobility_2
[0/8] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[7/8] cd /home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default/...Autopilot /home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: mobility_2
world: none
src_path: /home/ashok/Desktop/PX4-Autopilot
build_path: /home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/ashok/Desktop/PX4-Autopilot/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/ashok/px4_ros_com_ros2/install/ros2_aruco_interfaces/lib:/home/ashok/px4_ros_com_ros2/install/px4_ros_com/lib:/home/ashok/px4_ros_com_ros2/install/px4_msgs/lib:/home/ashok/px4_ros_com_ros2/install/custom_messages/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/opt/rviz_ogre_vendor/lib:/opt/ros/foxy/lib/x86_64-linux-gnu:/opt/ros/foxy/lib:/home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
Using: /home/ashok/Desktop/PX4-Autopilot/Tools/sitl_gazebo/models/mobility_2/mobility_2.sdf
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

gzserver not ready yet, trying again!
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.121.128
[Msg] Loading world file [/home/ashok/Desktop/PX4-Autopilot/Tools/sitl_gazebo/worlds/empty.world]
not running gazebo gui
SITL COMMAND: "/home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default"/etc -s etc/init.d-posix/rcS -t "/home/ashok/Desktop/PX4-Autopilot"/test_data
Creating symlink /home/ashok/Desktop/PX4-Autopilot/build/px4_sitl_default/etc -> /home/ashok/Desktop/PX4-Autopilot/buil

_____  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=1002
2022
INFO  [param] selected parameter default file eeprom/parameters_1002
etc/init.d-posix/rcS: 62: [: eeprom/parameters_1002: unexpected operator
[param] parameter file not found, creating eeprom/parameters_1002 2022
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 1002
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010

### Description
Command to access and manipulate parameters via shell or script.

This is used for example in the star
Usage: param <command> [arguments...]
 Commands:

   load          Load params from a file (overwrite all)
     [<file>   ] File name (use default if not given)

   import        Import params from a file
     [<file>   ] File name (use default if not given)

   save          Save params to a file
     [<file>   ] File name (use default if not given)

   dump          Dump params from a file
     [<file>   ] File name (use default if not given)

   select        Select default file
     [<file>   ] File name

   select-backup Select default file
     [<file>   ] File name

   show          Show parameter values
     [-a]        Show all parameters (not just used)
     [-c]        Show only changed params (unused too)
     [-q]        quiet mode, print only param value (name needs to be exact)
     [<filter> ] Filter by param name (wildcard at end allowed, eg. sys_*)

   show-for-airframe Show changed params for airframe config

   status        Print status of parameter system

   set           Set parameter to a value
     <param_name> <value> Parameter name and value to set
     [fail     ] If provided, let the command fail if param is not found

   set-default   Set parameter default to a value
     [-s]        If provided, silent errors if parameter doesn't exists
     <param_name> <value> Parameter name and value to set
     [fail     ] If provided, let the command fail if param is not found

   compare       Compare a param with a value. Command will succeed if equal
     [-s]        If provided, silent errors if parameter doesn't exists
     <param_name> <value> Parameter name and value to compare

   greater       Compare a param with a value. Command will succeed if param is greater than the value
     [-s]        If provided, silent errors if parameter doesn't exists
     <param_name> <value> Parameter name and value to compare
     <param_name> <value> Parameter name and value to compare

   touch         Mark a parameter as used
     [<param_name1> [<param_name2>]] Parameter name (one or more)

   reset         Reset only specified params to default
     [<param1> [<param2>]] Parameter names to reset (wildcard at end allowed)

   reset_all     Reset all params to default
     [<exclude1> [<exclude2>]] Do not reset matching params (wildcard at end allowed)

   index         Show param for a given index
     <index>     Index: an integer >= 0

   index_used    Show used param for a given index
     <index>     Index: an integer >= 0

   find          Show index of a param
     <param>     param name
ERROR [px4] Startup script returned with return value: 256
```

Jaeyoung-Lim, 2022-04-08:
@REGATTE As we discussed in https://discuss.px4.io/t/custom-uav-model-to-sitl/26861/10, this means you have a syntax error on loading your parameter

REGATTE, 2022-04-08:
@Jaeyoung-Lim 

Hi,
As mentioned earlier, I did check all parameters(both on the website and if they're available on the firmware) and their syntax.  

attached below is the param file

```
#! /bin/sh
#
# @name UAVTECH M2
#
# @type octo-coax-vtol
# @class logistics UAV
#
# @output MAIN1 - octo-coax motor1 			<input_index>0</input_index>
# @output MAIN2 - octo-coax motor2 			<input_index>1</input_index>
# @output MAIN3 - octo-coax motor3 			<input_index>2</input_index>
# @output MAIN4 - octo-coax motor4 			<input_index>3</input_index>
# @output MAIN5 - octo-coax motor5 			<input_index>4</input_index>
# @output MAIN6 - octo-coax motor6 			<input_index>5</input_index>
# @output MAIN7 - octo-coax motor7 			<input_index>6</input_index>
# @output MAIN8 - octo-coax motor8 			<input_index>7</input_index>
# @output AUX1 - THROTTLE RIGHT FRONT PULLER 		<input_index>8</input_index>
# @output AUX2 - THROTTLE LEFT FRONT PULLER  		<input_index>9</input_index>
# @output AUX3 - THROTTLE RIGHT BACK PUSHER 		<input_index>10</input_index>
# @output AUX4 - THROTTLE LEFT BACK PUSHER  		<input_index>11</input_index>
# @output AUX5 - AILERON RIGHT 				<input_index>12</input_index>
# @output AUX6 - AILERON LEFT	 			<input_index>13</input_index>
# @output AUX7 - ELEVATOR BACK	 			<input_index>14</input_index>
# @output AUX8 - PASS

# @maintainer Ashok Kumar J <ashok.kumar@uavtech.ai>
# @maintainer Aditya Alluri <aditya.alluri@uavtech.ai>
# @maintainer Siddartha Vadupu <siddartha.vadupu@uavtech.ai>
#
# @board px4_fmu-v5x only
# @board exclude all other
#

. ${R}etc/init.d/rc.vtol_defaults

# Param written for prototype, not for the actual model

# Add param below

# General Param
param set-default CA_AIRFRAME 2 # Airframe - 2: Standar VTOL
param set-default CA_ROTOR_COUNT 12 # total number of motors
param set-default CA_SV_CS_COUNT 3 #control surfaces: aileron right, aileron left, elevator
param set-default EKF2_HGT_MODE 0 # Determines the primary source of height data used by the EKF, 0: barometric pressure
param set-default PWM_AUX_RATE 400
param set-default PWM_AUX_MIN 1075
param set-default PWM_AUX_MAX 1950

# Transition refer https://docs.px4.io/master/en/advanced_config/parameter_reference.html#vtol-attitude-control
param set-default VT_F_TRANS_THR 0.75 # Target throttle value for the transition to fixed wing flight
param set-default VT_MOT_ID 12345678 # The channel number of motors which provide lift during hover
param set-default VT_FW_MOT_OFFID 12345678 # The channel number of motors that must be turned off in fixed wing mode
param set-default VT_TYPE 2 # standard vtol
# ADSB
param set-default NAV_TRAFF_AVOID 1 # ADSB traffic avoidance, 1: Warn mode, to warn via uORB messages

# RC Param
param set-default COM_RC_IN_MODE 0 # RC Transmitter only
param set-default COM_RC_LOSS_T 10 # Time in seconds after which loss of RC connection is flagged, only for testing the prototype
param set-default NAV_RCL_ACT 2 # RC failsafe mode, entered after COM_RC_LOSS_T, 2: Return mode, returns to TAKE OFF point

# satellite - gps params
param set-default EKF2_REQ_NSATS 12 # Required satellite count to use GPS
param set-deafult COM_ARM_WO_GPS 0 # Allow arming without GPS, 0 : cannot arm wo GPS
param set-deafult EKF2_GPS_MASK 1 # Blend multi-gps data w.r.t Horizontal Position as the main weight
# GPS Pos Transformation - add when model is created
# param set-default EKF2_GPS_POS_X
# param set-default EKF2_GPS_POS_Y
# param set-default EKF2_GPS_POS_Z
# look into EKF2_GPS_P_GATE, EKF2_GPS_P_NOISE, EKF2_GPS_TAU
param set-default GPS_1_CONFIG 201 # Serial Configuration for Main GPS
param set-default GPS_2_CONFIG 202 # Serial Configuration for Secondary GPS
param set-default GPS_DUMP_COMM 1 # Dump GPS communication to a file, all GPS data sent via uORB message, and written to log file as gps_dump
# param set-default GPS_YAM_OFFSET # Heading/Yaw offset for dual antenna GPS

param set-default BAT1_N_CELLS 4 #prototype battery - 4S

# Fixed Wing Speed Param Setting
param set-default FW_AIRSPD_MAX 33 # max airspeed 33m/s == 120kmph [max 40m/s == 144kmph]
param set-default FW_AIRSPD_MIN 20 # min airspeed 20m/s == 72kmph [stall speed = 70kmph]
param set-default FW_ARSPD_TRIM 27.5 # cruise air speed 27.5 == 99kmph

# MultiCopter Speed Patam setting
param set-default MPC_ACC_UP_MAX 5 # Maximum vertical acceleration 5m/s == 18kmph
#param set-default MPC_THR_HOVER  # Vertical thrust required to hover. (mapped between 0.1 - 0.8, increment = 0.01, default 0.5)

# MultiCopter landing
# 2 STEP LANDING
param set-default MPC_LAND_ALT1 50 #Entering landing phase-2, Below this altitude, descending velocity gets limited to a value between "MPC_Z_VEL_MAX" & "MPC_LAND_SPEED"
param set-default MPC_LAND_ALT2 20 # Entering Landing phase-3, below this altitude descedning velocity is limited to "MPC_LAND_SPEED".
# param set-default MPC_Z_VEL_MAX 1.5m/s
# param set-default MPC_LAND_SPEED 0.5m/s

# end param

# SITL/HITL FUNCTIONALITY
#param set SYS_HITL 1

# disable some checks to allow to fly
# - with usb
#param set-default CBRK_USB_CHK 197848
# - without real battery
#param set-default CBRK_SUPPLY_CHK 894281
# - without safety switch
#param set-default COM_PREARM_MODE 0
#param set-default CBRK_IO_SAFETY 22027

set MAV_TYPE 22 # setting on VTOL Reserved 2 point as its a custom frame. ref: https://mavlink.io/en/messages/common.html#MAV_TYPE

set MIXER octo_m2_h #mixer defined, toml defined
# defining aux connections in .post file
# set MIXER_AUX m2_plane #mixer defined, toml to be defined

# Need to set all channels
set PWM_OUT 12345678
# set PWM_AUX_OUT 1234567
```

Jaeyoung-Lim, 2022-04-08:
@REGATTE which line is this message being triggered from your airframe startup script?

```

### Description
Command to access and manipulate parameters via shell or script.

This is used for example in the star
Usage: param <command> [arguments...]
 Commands:

```

Jaeyoung-Lim, 2022-04-11:
@REGATTE Any updates?

REGATTE, 2022-04-16:
hi @Jaeyoung-Lim, 
sorry for the late reply. I was out on a holiday.

I'm sorry had a couple of typos's all over, which I didn't catch. 

The model is now running on the gazebo, but this error pops us
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [init] found model autostart file as SYS_AUTOSTART=2023
INFO  [param] selected parameter default file eeprom/parameters_2023
[param] parameter file not found, creating eeprom/parameters_2023
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 2023
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
* CAL_MAG0_ID: curr: 0 -> new: 197388
* CAL_MAG1_ID: curr: 0 -> new: 197644
* SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
* SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
* SYS_AUTOCONFIG: curr: 1 -> new: 0
* IMU_INTEG_RATE: curr: 200 -> new: 250
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
PX4 SIM HOST: localhost
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [commander] LED: open /dev/led0 failed (22)
* PWM_AUX_OUT: curr: 0 -> new: 1234
* PWM_MAIN_OUT: curr: 0 -> new: 1234
INFO  [init] Mixer: etc/mixers/octo_m2_h.main.mix on /dev/pwm_output0
INFO  [init] setting PWM_AUX_OUT none
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.121.129
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-04-16/14_55_59.ulg	
INFO  [logger] Opened full log file: ./log/2022-04-16/14_55_59.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [mavlink] MAVLink only on localhost (set param MAV_{i}_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [tone_alarm] home set
WARN  [mc_pos_control] invalid setpoints
INFO  [tone_alarm] notify negative
[Wrn] [Publisher.cc:135] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.
INFO  [mavlink] partner IP: 127.0.0.1
pxh> commander arm -f
pxh> INFO  [commander] Armed by internal command	
INFO  [tone_alarm] arming warning
WARN  [ekf2] primary EKF changed 0 (accel fault) -> 2
WARN  [commander] Imbalanced propeller detected, landing advised
WARN  [ekf2] primary EKF changed 2 (filter fault) -> 0
INFO  [commander] Takeoff detected	
INFO  [commander] Landing detected	
INFO  [commander] Takeoff detected	
pxh> cINFO  [commander] Landing detected
INFO  [commander] Takeoff detected	
WARN  [vehicle_imu] Accel 1 clipping, not safe to fly!	
WARN  [vehicle_imu] Accel 2 clipping, not safe to fly!	
pxh> commander takeoff
pxh> INFO  [navigator] Using minimum takeoff altitude: 2.50 m	
WARN  [ekf2] primary EKF changed 0 (accel fault) -> 2
WARN  [ekf2] primary EKF changed 2 (filter fault) -> 0
WARN  [vehicle_imu] Accel 1 clipping, not safe to fly!	
WARN  [vehicle_imu] Accel 2 clipping, not safe to fly!	

[1]+  Stopped                 make px4_sitl gazebo_mobilityVtol_2

```

the main things are
```
WARN  [commander] Imbalanced propeller detected, landing advised
```

Jaeyoung-Lim, 2022-04-16:
@REGATTE This means either you have something out of balance (e.g. rotors spinning with the cg out of axis), or your imus are vibrating due to some link misttached to the base link.

I think this is coming from your model and no longer an issue on the px4 side

REGATTE, 2022-04-16:
oh okay,
thank you.

Is it possible to know which link/joint is causing an issue??

Jaeyoung-Lim, 2022-04-16:
@REGATTE Since you never shared the proprietary model, it is hard to know even which links/joints exists :stuck_out_tongue_winking_eye: 

I would try to compare your model with the default model and see if some things are differnet. e.g. you are connecting the imu in a different way .etc

Jaeyoung-Lim, 2022-04-22:
@REGATTE As I mentioned, nothing is still in gazebo since the dynamics engine is constantly solving contacts (stationary models are in collision with a ground plan). I would check if this happens on the default models and if not you need to find out what is wrong with your custom model.

Jaeyoung-Lim, 2022-08-18:
Closing as stale, please feel free to reopen if it is relevant

closed 2022-08-18 by Jaeyoung-Lim