PX4/PX4-Autopilot #19459: [Project Tracker] Sensor Calibration Improvement for PX4

junwoo091400, 2022-04-08:
## Why
Currently, Sensor calibrations are really tedious, especially the 6 point magnetometer calibration dance you need to do in the field. This is cumbersome and inconvenient. Also, whenever you calibrate a sensor, you need to specify the mounting orientation, which is really confusing (e.g. Calibrating the Internal Magnetometer requires you to set the orientation in QGC, why?).

Sensor Calibrations should become much simpler.

## Goal of the Project
1. **Remove 6 point Mag dance & Acc calibration routine, or dramatically simplify it**
2. Unify Sensor Calibration so that user can just push one button and **calibrate everything at once**
3. **Clearly define Sensor orientation rules** (Is it in reference to the Board? the Gravity? etc)
4. Make Automatic Sensor Orientation detection work seamlessly ([Currently it can return false values](https://discuss.px4.io/t/px4-dev-call-april-06-2022/26787#sensor-calibration-improvements-junwoo0914-12))
6. **Allow arbitrary sensor mount orientation** (No hard rules like Roll 0, Pitch 270, etc.)

## TODOs
- [ ] Understand current sensor calibration architecture
- [ ] Discuss ideas on how to achieve these goals above, with stakeholders
- [ ]

chris1seto, 2022-04-08:
One comment: For sensor alignment, the presets are actually helpful because the alternative is having to manually input a rotation matrix, which might be confusing for users. Perhaps there could be some presets in qgc or just the ability to enter a raw quat. The actual param might be a quat, I'm just suggesting covering it in the front end unless the user wants to specify it themselves.

ryanjAA, 2022-04-08:
@junwoo091400 this is so very needed. 👍👍👍

junwoo091400, 2022-04-11:
Suggestion from @chris1seto from [Slack](https://px4.slack.com/archives/C50RWGQMT/p1649430999304469?thread_ts=1649421934.662239&cid=C50RWGQMT)

> One suggestion I would have is for there to be an easy way to export/import calibration parameters only. I realize this might be more of a qgc than Px4 feature, but the **single worst part of doing a parameter reset for me is redoing calibration because I don't have a way of only reloading calibration params only.** I realize there might be some editing of a normal parameter export that's possible, I just mean that there is no standard way of doing this.

junwoo091400, 2022-04-11:
Input from @cmic0 :
1. Would be nice to have a voice saying "stop", or "turn", when doing calibration, not just a diagram showing what to do
2. Change the diagram from Fixed Wing to Copter :thinking: 
3. We only need to do 6 point mag dance once, so it's not really cumbersome.

junwoo091400, 2022-04-26:
Added 2 TODOS from the last dev call : https://discuss.px4.io/t/px4-dev-call-april-20-2022/26970

ryanjAA, 2022-04-26:
Option for copter (checkbox?) or fixed wing icon since both are used makes more sense than one or the other since both are used and relevant.

junwoo091400, 2022-04-29:
> One comment: For sensor alignment, the presets are actually helpful because the alternative is having to manually input a rotation matrix, which might be confusing for users.

@chris1seto What do you mean by 'preset'? The ROTATION parameter of the sensors? 🤔

closed 2025-10-29 by github-actions[bot]