PX4/PX4-Autopilot #19462: Using rangefinders for fuel or liquid level measurements (UAVCAN)

dirksavage88, 2022-04-08:
What fuel tank status sensors have been commonly used for generic fuel tank messages? I have been looking at using a cheap ToF distance sensor for 'fuel' level sensing [example](https://youtu.be/yytfrtn2pmc)

 I see there is a fuel tank msg in the DSDL [here](https://github.com/dronecan/DSDL/blob/35c593c6b1a6eefb3310b2c14121d9b66710ecfa/uavcan/equipment/ice/1129.FuelTankStatus.uavcan#L18 )

It looks feasible to take rangefinder measurements and do some simple calculations of percentage of fuel based on fuel tank characteristics (as can node publisher parameters). There is also a possibility to wrap this, along with the rangefinder orientations, into the sensor module?

adk4davsys, 2022-04-11:
Hello @dirksavage88,

I think this question is better suited for drone forums as this is not directly related to PX4 issue.

dagar, 2022-04-11:
Ideally the first choice would be a "real" fuel level sensor that we interface with in its native protocol. 

If that's not a realistic option for whatever reason then we could add a driver/module within PX4 that consumes the corresponding raw data + user supplied configuration parameters and publishes a fuel tank estimate. You could even try to factor in the attitude estimate or other data to deal with tank sloshing. 

If it makes sense architecturally you could then run this on a CAN node that publishes FuelTankStatus.

closed 2024-05-30 by dirksavage88