PX4/PX4-Autopilot #19463: px4_fmu_v5x uavcan Unable to start self startup

zhnliuliu, 2022-04-09:
**Describe the bug**
param uavcan_Enable has been set to 3, but the uavcan service will not start automatically

**To Reproduce**
Steps to reproduce the behavior:
1. Power pixhawk FMU v5x through USB
2. start uavcan error by checking the debug serial port information

**Expected behavior**
Prompt that the uavcan fails to start. Use the mavlink console command: 
`uavcan status`

**Log Files and Screenshots**
[image]

zhnliuliu, 2022-04-09:
This problem occurs on boards that do not use px4io

zhnliuliu, 2022-04-09:
I don't know if px4io will appear again after it is enabled

dagar, 2022-04-09:
Do you have an SD card in?

zhnliuliu, 2022-04-09:
SD card, of course

zhnliuliu, 2022-04-09:
Manual start cannot be started, and the uavcan start command must be run twice in a row

dagar, 2022-04-10:
> Manual start cannot be started, and the uavcan start command must be run twice in a row

I'm not sure what you mean, can you share the full output? Which version of PX4 is this?

zhnliuliu, 2022-04-10:
master branch

zhnliuliu, 2022-04-10:
http://px4-travis.s3.amazonaws.com/Firmware/master/px4_fmu-v5x_default.px4

zhnliuliu, 2022-04-10:
HW arch: PX4_FMU_V5X
HW type: V5X02
HW version: 0x00000000
HW revision: 0x00000002
FW git-hash: 091fca701e964de228ea077e11053119104d94fc
FW version: 1.13.0 40 (17629248)
OS: NuttX
OS version: Release 10.2.0 (167903487)
OS git-hash: c5c7d2b4f26f52f1dfb425ebde83328606b65d4f
Build datetime: Apr  9 2022 19:52:40
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 00020000000020333537594250100031001c
MCU: STM32F76xxx, rev. Z
INFO  [param] selected parameter default file /fs/mtd_params
INFO  [param] importing from '/fs/mtd_params'
WARN  [parameters] ignoring unrecognised parameter 'CAL_GYRO0_TEMP'
WARN  [parameters] ignoring unrecognised parameter 'CAL_GYRO1_TEMP'
WARN  [parameters] ignoring unrecognised parameter 'CAL_GYRO2_TEMP'
INFO  [parameters] BSON document size 2306 bytes, decoded 2306 bytes (INT32:28, FLOAT:87)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
INFO  [dataman] data manager file '/fs/microsd/dataman' size is 62560 bytes
Loading airframe: /etc/init.d/airframes/4001_quad_x
INFO  [uavcan] Node ID 1, bitrate 1000000
ERROR [uavcan] CAN driver init failed -1006
WARN  [px4io] check CRC failed: -1, CRC: 1043691576
WARN  [px4io] IO refused to re-enable safety
WARN  [px4io] IO refused to reboot
WARN  [px4io] reboot failed - -1, still attempting upgrade
[PX4IO] using firmware from /etc/extras/px4_io-v2_default.bin
[PX4IO] bootloader not responding
WARN  [px4io] IO refused to re-enable safety
WARN  [px4io] IO refused to reboot
WARN  [px4io] reboot failed - -1, still attempting upgrade
[PX4IO] using firmware from /etc/extras/px4_io-v2_default.bin
[PX4IO] bootloader not responding
ERROR [px4io] error updating PX4IO - check that bootloader mode is enabled
INFO  [tune_control] Stopping playback...
PX4IO not found
Board sensors: /etc/init.d/rc.board_sensors
INFO  [ina226] Failed to init INA226 on bus 1, but will try again periodically.
ina226 #0 on I2C bus 1 (external) address 0x41
INFO  [ina226] Failed to init INA226 on bus 2, but will try again periodically.
ina226 #1 on I2C bus 2 (external) address 0x41
icm20649 #0 on SPI bus 3 rotation 6
icm42688p #0 on SPI bus 2 rotation 6
icm20602 #0 on SPI bus 1 rotation 10
bmm150 #0 on I2C bus 4 address 0x10
WARN  [SPI_I2C] ist8310: no instance started (no device on bus?)
bmp388 #0 on I2C bus 4 address 0x77
bmp388 #1 on I2C bus 2 (external) address 0x76
ERROR [mixer] can't open /dev/pwm_output0

ERROR [mixer] failed to load mixer
ERROR  [init] Failed loading mixer: /etc/mixers/quad_x.main.mix
INFO  [init] setting PWM_AUX_OUT none
ekf2 [1083:237]
Board mavlink: /etc/init.d/rc.board_mavlink
Starting Main GPS on /dev/ttyS0
Starting MAVLink on /dev/ttyS6
WARN  [mavlink] offboard mission init failed (-1)
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS6 @ 57600B
Starting MAVLink on ethernet
INFO  [mavlink] mode: Normal, data rate: 100000 B/s on udp port 14550 remote port 14550
INFO  [logger] logger started (mode=all)

NuttShell (NSH) NuttX-10.2.0
nsh> [KWARN  [mavlink] no broadcasting address found
INFO  [commander] Failsafe mode activated	
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] /fs/microsd/log/2022-04-10/09_06_45.ulg	
ERROR [logger] Can't open log file /fs/microsd/log/2022-04-10/09_06_45.ulg, errno: 20
INFO  [mavlink] using network interface eth0, IP: 192.168.0.3
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 192.168.0.255

zhnliuliu, 2022-04-10:
The full log is here：[https://logs.px4.io/plot_app?log=87e7b778-eb3e-4e59-a394-5fa088be64d1](https://logs.px4.io/plot_app?log=87e7b778-eb3e-4e59-a394-5fa088be64d1)

zhnliuliu, 2022-04-10:
UAVCAN_ENABLE, 2 but uavcan not started

[image]

zhnliuliu, 2022-04-10:
> > Manual start cannot be started, and the uavcan start command must be run twice in a row
> 
> I'm not sure what you mean, can you share the full output? Which version of PX4 is this?

hi，dagar The problem has been solved！！
The problem is caused by my replacement of IO board，Replace with the official IO board to start normally

zhnliuliu, 2022-04-10:
However, this problem should be solved. There is only one can interface on the unofficial IO board, which leads to initialization failure

zhnliuliu, 2022-04-10:
In addition, the SD card must be replaced before it can be started normally. The original SD card cannot start the service

closed 2025-10-29 by github-actions[bot]