PX4/PX4-Autopilot #19471: Question about developer mode

its-pabs, 2022-04-11:
Hi there,

I have been building an X8 Cine drone to lift around 35lbs AUW. Ive been having some trouble tuning, however, as the herelink Ground station's QGC is very limited, as it only has 2 sliders for tuning, one for hover and one for idle throttle. 

My next step in this project is to use the autotune feature to tune my drone. I have put it into developer mode for this. Id like to know how safe that is, what I should be aware of, etc since I was warned when I booted it to my flight controller that it has not been flight tested.

Thank you

bresch, 2022-04-12:
Hi @its-pabs ,
I would first suggest to update your QGC version in order to get the new PID tuning tab: 
https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter_basic.html

Also make sure to have the gyro filters set properly to get the optimal noise and lag for your setup:
https://docs.px4.io/master/en/config_mc/filter_tuning.html

About autotune, it's safe if you follow carefully what's written in the guide:
https://docs.px4.io/master/en/config/autotune.html
Just be careful, when the new gains are applied, takeoff slowly in stabilized to test the tuning, don't go into position or mission mode directly.

Please post your questions that are not bug reports on the forum instead: http://discuss.px4.io

closed 2022-04-12 by bresch

its-pabs, 2022-04-12:
Hi matthiew,

Thank you for your response, that is definitely very helpful. As for QGC, I have the new pid tuning tabs on my computer but not on my herelink ground station. Is this QGC version available for the herelink GS?

Thank you,
Pablo Villa

Professional Videographer & Photographer
Purdue Athletics Creative Intern
Filming, Editing, Color Grading, & Audio
Portrait and Group Photography
pablovillavideo.com<http://pablovillavideo.com/>
203-326-0150
***@***.******@***.***>

https://docs.px4.io/master/en/config_mc/pid_tuning_guide_multicopter_basic.html

https://docs.px4.io/master/en/config_mc/filter_tuning.html

https://docs.px4.io/master/en/config/autotune.html