PX4/PX4-Autopilot #19480: mRo H7 OEM New, telemetry doesn’t work

tomcats60, 2022-04-13:
I'm using the latest PX4 version on the mRo H7 OEM NEW with the reference design carrier board (https://store.mrobotics.io/product-p/m10066a.htm). 
QGC detects the FC by USB but not through telemetry even if both telemetry module are linked (fixed yellow LED)

I tried on TELEM1 (UART2 = /dev/ttyS0) and TELEM2 (UART3 = /dev/ttyS1) but result is the same.

taileron, 2022-04-13:
with the Stick Adapter
https://store.mrobotics.io/product-p/m10062a.htm
both telemetry ports (close to USB) work perfect with ESP 8266

ryanjAA, 2022-04-14:
Depending on when you received the board there was a batch with some hardware issues iirc. 

See here (https://px4.slack.com/archives/C0W2KUFFT/p1648319405997709?thread_ts=1648302068.722429&channel=C0W2KUFFT&message_ts=1648319405.997709)

closed 2025-10-29 by github-actions[bot]