PX4/PX4-Autopilot #19488: How to I get the raw sample timestamp for the imu data sensorcombined

stevedanomodolor, 2022-04-14:
I am trying to a vio with the imu reading in the ros2 interface with PX4 with the topic sensorcombined. Which is the timestamp of the imu data. I am working in simulation. Thank you. I have tried using the timestamp but does not seems to works. Is there a way to convert this to a ros time for example. I did this in my code

```
void ROS2Visualizer::callback_inertial(const px4_msgs::msg::SensorCombined::UniquePtr msg) {

  // convert into correct format
  ov_core::ImuData message;
  // message.timestamp = msg->header.stamp.sec + msg->header.stamp.nanosec * 1e-9;
  // message.wm << msg->angular_velocity.x, msg->angular_velocity.y, msg->angular_velocity.z;
  // message.am << msg->linear_acceleration.x, msg->linear_acceleration.y, msg->linear_acceleration.z;
```

```
 Sensor readings in SI-unit form.
# These fields are scaled and offset-compensated where possible and do not
# change with board revisions and sensor updates.

uint64 timestamp				# time since system start (microseconds)

int32 RELATIVE_TIMESTAMP_INVALID = 2147483647	# (0x7fffffff) If one of the relative timestamps is set to this value, it means the associated sensor values are invalid

# gyro timstamp is equal to the timestamp of the message
float32[3] gyro_rad			# average angular rate measured in the FRD body frame XYZ-axis in rad/s over the last gyro sampling period
uint32 gyro_integral_dt		# gyro measurement sampling period in microseconds

int32 accelerometer_timestamp_relative	# timestamp + accelerometer_timestamp_relative = Accelerometer timestamp
float32[3] accelerometer_m_s2		# average value acceleration measured in the FRD body frame XYZ-axis in m/s^2 over the last accelerometer sampling period
uint32 accelerometer_integral_dt	# accelerometer measurement sampling period in microseconds

uint8 CLIPPING_X = 1
uint8 CLIPPING_Y = 2
uint8 CLIPPING_Z = 4
uint8 accelerometer_clipping            # bitfield indicating if there was any accelerometer clipping (per axis) during the sampling period

uint8 accel_calibration_count  	# Calibration changed counter. Monotonically increases whenever accelermeter calibration changes.
uint8 gyro_calibration_count   	# Calibration changed counter. Monotonically increases whenever rate gyro calibration changes.
```

dagar, 2022-04-14:
I would use the `timestamp_sample` field from vehicle_imu (published per IMU instance). 

https://github.com/PX4/PX4-Autopilot/blob/6e0ac66c3c11872392c33a2bc3fd5fbbfcc6a3f8/msg/vehicle_imu.msg#L4

At some point `sensor_combined` will be dropped entirely.

dagar, 2022-04-14:
Sorry, just realized you're talking more about the ROS2 offset handling.

>  I have tried using the timestamp but does not seems to works

Can you expand on this? What are you actually seeing in the data?

stevedanomodolor, 2022-04-14:
Tge timestamp in the sensorcombined message. I will try the vehicle imu and see if it improves.

anishgollakota, 2024-02-19:
Any update on this?

closed 2025-10-29 by github-actions[bot]