PX4/PX4-Autopilot #19499: [ros2&gazebo]  sitl shows 'BARO #0 failed' with gazebo by ros2

LiuWhale, 2022-04-18:
Environment: 
- Matlab on win11
- ubuntu20.04LTS VM on win11
- ros2 foxy (configed by following the px4.io page)
- Firmware latest, master branch

Steps:

- add some ros2 msg: actuator_servosactuator_outputs actuator_armedactuator_motorsactuator_controls actuator_controls_0 actuator_controls rc_channels manual_control_setpoint manual_control_input
- regenerate yaml to workspace and rebuild firmware and workspace
- write sh to launch sitl:
```
#!/bin/bash

export ROS_VERSION=2
export ROS_DISTRO=foxy
export PX4AUTOPILOTHOME=/home/user/ros2_ws/PX4-Autopilot
export GAZEBO_PLUGIN_PATH=${GAZEBO_PLUGIN_PATH}:$PX4AUTOPILOTHOME/build
export GAZEBO_MODEL_PATH=${GAZEBO_MODEL_PATH}:$PX4AUTOPILOTHOME/Tools/sitl_gazebo/models
export GAZEBO_MODEL_DATABASE_URI=""
export SITL_GAZEBO_PATH=$PX4AUTOPILOTHOME/Tools/sitl_gazebo

gnome-terminal --title="PX4 SITL RTPS Gazebo" -- /bin/bash -c 'source ~/px4_ros_com_ros2/install/setup.bash; cd $PX4AUTOPILOTHOME; make px4_sitl_rtps gazebo_iris '

sleep 12

gnome-terminal --title="microRTPS Agents" -- /bin/bash -c 'source ~/px4_ros_com_ros2/install/setup.bash; micrortps_agent -t UDP -r 2020 -s 2019 '
```
- write matlab code:
```
clc;clear;
node = ros2node("/control_node");

controlModePub = ros2publisher(node,"fmu/offboard_control_mode/in","px4_msgs/OffboardControlMode");
timeSub = ros2subscriber(node,"fmu/timesync/out","px4_msgs/Timesync");

cmdPub = ros2publisher(node,"/fmu/vehicle_command/in","px4_msgs/VehicleCommand");
ac0Pub = ros2publisher(node,"/fmu/actuator_controls0/in","px4_msgs/ActuatorControls0"); 

timeLimit = 5;
receive(timeSub,timeLimit);

tStart = tic;
control = [0 0 0 1 0 0 0 0]';
while toc(tStart) < 1
    arm(timeSub,cmdPub)
    publishOffboardControlMode(timeSub,controlModePub,"actuator")
    publishActuatorControls0(timeSub,ac0Pub,control)
    pause(0.1)
end

engageOffboardMode(timeSub,cmdPub)
arm(timeSub,cmdPub)
while toc(tStart) < 40
    publishOffboardControlMode(timeSub,controlModePub,"actuator")
    publishActuatorControls0(timeSub,ac0Pub,control)
end
% Land once complete
land(timeSub,cmdPub)
```

Result:
[image]

Jaeyoung-Lim, 2022-04-19:
@LiuWhale Offboard actuator setpoints are not supported with SITL since the actuator controls signals are used for lockstep

LiuWhale, 2022-04-19:
> 

Thank you for reply. So, does it have any solutions to control servo derectly in offboard mode with ros2?

Jaeyoung-Lim, 2022-04-19:
@LiuWhale This is not related to ROS2, and it will work on a real vehicle.

This is just a limitation of SITL simulations using lockstep (e.g. gazebo) You can turn off lockstep, but the performance might be bad id you are not careful with the configuration

LiuWhale, 2022-04-20:
@Jaeyoung-Lim Fine~ I have tried to disable lockstep and the performance was same as you said. Maybe it's time to have real test. hhha

closed 2022-04-20 by LiuWhale