PX4/PX4-Autopilot #19503: KakuteH7 firmware very slow on startup

sstroobants, 2022-04-19:
When flashing the KakuteH7 with the latest master build, the FC boots very very slowly (taking up to 20 seconds). The default Betaflight firmware does not have this issue. I've also tested removing a lot of the modules to see if that would change things, but it didn't. Because of this, the ESCs do not get the proper commands on time and thus never finish the startup sequence. Only way now to use it is to power the FC first by usb cable, wait until it has completely booted and only then power the ESCs.

**To Reproduce**
1. Build the KakuteH7 bootloader (latest master) and flash it using `dfu-util`.
2. Flash the custom master build using QGroundControl
3. Now the FC takes about 20 seconds to boot, which if using the batteries results in only the first three beeps of the ESCs, but not the two last ones.

**Drone:**
- Custom drone with the KakuteH7 FC and Tekko32.

dagar, 2022-04-19:
@bkueng

bkueng, 2022-04-20:
Yes I noticed that as well (https://github.com/PX4/PX4-Autopilot/pull/19019). Seems to be something lower-level, and I don't have a jtag setup for the board.

sstroobants, 2022-04-20:
Ah I see, I missed that line in that commit. Is there any way I can help looking into this? Or do you have a clue why the ESCs do not start up after this extremely long boot time? Do they go into some kind of standby mode or something when they don't receive the throttle low/high values fast enough? Is there any way to get around that?

sstroobants, 2022-08-09:
Hi, I see that in the latest stable release, this is still an issue. Could that be correct? So still the ESC's are not initializing after starting up and it takes around +-20 seconds for startup. Let me know if there exists a fix.

bkueng, 2022-08-10:
Yes that is correct, it has not been fixed.

sstroobants, 2022-08-10:
@bkueng, are there any plans for fixing this in the near future? Is there any way in which I could be of help?

bkueng, 2022-08-11:
I probably don't have the time, but you're very welcome to look into this. I'd hook up a debugger to check where in the bootup it hangs.

sstroobants, 2022-08-19:
Okay I have attached a black magic probe to the SWD pins of the KakuteH7 and am able to debug using VS Code @bkueng. One issue I have now is that, although I can manually pause, the software does not seem to stop at breakpoint that I set. Do you have any clue why that might be the case? This is making it more difficult for me to find out where it is hanging in the bootup.

bkueng, 2022-08-19:
Is this on latest main? In the NuttX source code? Does it work if you set one in a PX4 source file?

sstroobants, 2022-08-19:
Yes on latest main and in the NuttX source code. It also doesn't work in a PX4 source file. I have been able to set a breakpoint when I specifically set a hardware breakpoint when I use gdb in the console however, so it seems like it is an issue with how I define the breakpoint in VS Code, maybe I am missing something there?

bkueng, 2022-08-19:
Looks like it. I'm not using VS code, so I can't tell.

dagar, 2022-08-19:
@sstroobants in vscode did you select the kakuteh7 variant (for cmake configuration)? I use vscode debugging fairly often on various boards, but with a jlink. It could be the blackmagic config within vscode that's not configured properly.

sstroobants, 2022-08-22:
Yes I did use the kakuteh7 variant (it was not in the cmake-variants.yaml file, but I added it myself:
```    
holybro_kakuteh7_default:
      short: holybro_kakuteh7
      buildType: MinSizeRel
      settings:
        CONFIG: holybro_kakuteh7_default
```
I can have another look at the blackmagic config but I am not sure what I could change there to specifically set the hardware breakpoints.

vincentpoont2, 2024-03-29:
@julianoes @bkueng Any idea why there is such a delay? Doesn't happen in betaflight. 

https://github.com/PX4/PX4-Autopilot/assets/46874772/580fcc1c-0c21-410f-9d96-45e96ec0e37b

julianoes, 2024-04-01:
Is this more than just the 5s of the bootloader? I believe the bootloader USB connected check doesn't work correctly leading to a 5s delay.

vincentpoont2, 2024-04-01:
> Is this more than just the 5s of the bootloader? I believe the bootloader USB connected check doesn't work correctly leading to a 5s delay.

It seems to take around 20 seconds

julianoes, 2024-04-01:
I'll try to reproduce. Best would be to get the output of `dmesg` after this happens.

sstroobants, 2024-05-17:
[param_import_fail.txt](https://github.com/PX4/PX4-Autopilot/files/15349091/param_import_fail.txt)

for what it's worth; this file gets created at boot. And the shell constantly prints: 
`ERROR [parameters] param export failed (-27)`
`INFO  [parameters] param auto save unavailable (-27), retrying..`

sstroobants, 2024-05-17:
> [param_import_fail.txt](https://github.com/PX4/PX4-Autopilot/files/15349091/param_import_fail.txt)
> 
> for what it's worth; this file gets created at boot. And the shell constantly prints: `ERROR [parameters] param export failed (-27)` `INFO [parameters] param auto save unavailable (-27), retrying..`

This works temporarily by commenting out: 
`#define FLASH_BASED_PARAMS`
in /PX4-Autopilot/boards/holybro/kakuteh7/src/board_config.h

But after a restart the KakuteH7 ends up in a booting loop, for some reason.

sstroobants, 2024-05-17:
The most annoying thing is that the ESC's don't turn on anymore if they don't receive the command from the FC fast enough. I think that is quite weird actually, and I would be happy with the situation if that gets fixed, and I just have to wait a little bit longer before the whole quad is fully booted.

sstroobants, 2024-05-17:
> @julianoes @bkueng Any idea why there is such a delay? Doesn't happen in betaflight.
> VID_20240326_120327_123.mp4.mov

Interesting that in this example the ESCs only start powering up when the FC is already powered up. Do you know why that might be the case?

azerupi, 2025-02-19:
This issue is still present in main (assuming it is the same).

When connecting to the USB on the Kakute and looking at `dmesg` logs from my computer we can see the following:

- The bootloader starts to run 
- 3 seconds later the bootloader disconnected from USB, supposedly because it jumped to the application.
- The application USB only comes on 21 seconds after boot or 18 seconds after the bootloader supposedly jumped to the application.

For comparison, the Pixhawk 6X boots in 5 seconds for the bootloader and 2 seconds for the app.

**dmesg logs on my computer**
```
[763283.168425] usb 3-4.3.4.3: new full-speed USB device number 10 using xhci_hcd
[763283.260628] usb 3-4.3.4.3: New USB device found, idVendor=3162, idProduct=0050, bcdDevice= 1.01
[763283.260646] usb 3-4.3.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[763283.260651] usb 3-4.3.4.3: Product: PX4 BL Holybro KakuteH7
[763283.260654] usb 3-4.3.4.3: Manufacturer: Holybro
[763283.260657] usb 3-4.3.4.3: SerialNumber: 0
[763283.269882] cdc_acm 3-4.3.4.3:1.0: ttyACM0: USB ACM device

[763286.056712] usb 3-4.3.4.3: USB disconnect, device number 10

[763304.676353] usb 3-4.3.4.3: new full-speed USB device number 11 using xhci_hcd
[763304.767520] usb 3-4.3.4.3: New USB device found, idVendor=3162, idProduct=0050, bcdDevice= 1.01
[763304.767526] usb 3-4.3.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[763304.767529] usb 3-4.3.4.3: Product: PX4 KakuteH7
[763304.767530] usb 3-4.3.4.3: Manufacturer: Holybro
[763304.767531] usb 3-4.3.4.3: SerialNumber: 0
[763304.776685] cdc_acm 3-4.3.4.3:1.0: ttyACM0: USB ACM device
```

On the console, I'm not getting any output for 20 seconds as well. The console output appears at the same time than the ttyACM.

**console output**
```
[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
ERROR [PX4_MTD] failed to locate spi bus
ERROR [PX4_MTD] mtd failure: -6 bus 2 address 0 class 1
HW arc[hardfault_log] Fault Log is Armed
h: HOLYBRO_KAKUTEH7
PX4 git-hash: 01549a58326c47cf562dc190f7fa6eaf966516cc
PX4 version: 1.16.0 40 (17825856)
PX4 git-branch: kakute
OS: NuttX
OS version: Release 11.0.0 (184549631)
OS git-hash: 9a213327fbb2dab252185d468fb29348e1833d8c
Build datetime: Feb 17 2025 11:17:17
Build uri: localhost
Build variant: default
Toolchain: GNU GCC, 9.3.1 20200408 (release)
PX4GUID: 000600000000313436303133510f004f0037
MCU: STM32H7[4|5]xxx, rev. V
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
New /fs/mtd_caldata size is:
ERROR [bsondump] open '/fs/mtd_caldata' failed (2)
INFO  [param] selected parameter backup file /fs/microsd/parameters_backup.bson
Board architecture defaults: /etc/init.d/rc.board_arch_defaults
Board defaults: /etc/init.d/rc.board_defaults
Loading airframe: /etc/init.d/airframes/4050_generic_250
WARN  [dataman] Could not seek data manager file /fs/microsd/dataman
ERROR [dataman] dataman start failed
Board sensors: /etc/init.d/rc.board_sensors
mpu6000 #0 on SPI bus 4 rotation 6
bmp280 #0 on I2C bus 1 address 0x76
nsh: icm20948_i2c_passthrough: command not found
nsh: lis2mdl: command not found
qmc5883l #0 on I2C bus 1 address 0xD
ekf2 [598:237]
nsh: px4io: command not found
Starting Main GPS on /dev/ttyS3
Starting RC Input Driver on /dev/ttyS4
INFO  [cdcacm_autostart] Starting CDC/ACM autostart
Board extras: /etc/init.d/rc.board_extras
ERROR [param] Parameter OSD_ATXXXX_CFG not found
nsh: atxxxx: command not found
INFO  [logger] logger started (mode=all)
```

At some point, after erasing the flash and flashing with JLink I managed to get the Kakute in a good state again. And it was booting in <10s. But then one update through QGC triggered the problem again.

I am however unable to repeat this at this point, I'm not sure which combination of actions put the Kakute in a good state. I'm just mentioning all of this in case it triggers any ideas for someone else.

julianoes, 2025-02-19:
There is one issue that I see consistently which is that it always seems to go through the bootloader somehow, so I'm going to look into that.

Also, there was a problem with param erase (which is why the dfu mass erase was required). Fix is coming: https://github.com/PX4/PX4-Autopilot/pull/24371

julianoes, 2025-03-18:
Slow start fixed in #24520.

closed 2025-03-18 by julianoes