PX4/PX4-Autopilot #19506: MPU9250 not found after v1.12.0 on BeagleBone blue

navunlimited, 2022-04-20:
**Describe the bug**
I was able to make MPU9250 work ok on v1.11.3. But here are other issues about pwm test and esc calibration since bblue pwm mapping is different from PX4 default. So I tried to update to newer version v1.12.3 since there are new board_pwm_out files there. After recompile and upload the v1.12.3, MPU9250 cannot be found at all. So, I tried v1.12.0. The issue is the same. It seems that PX4-Autopilot/src/drivers/imu/mpu9250 directory is removed or combined into invensense/mpu9250 after v1.12.0. But something goes wrong.
PS. pwm functions seem not working as well since escs keep beeping after lunch px4.....   

**To Reproduce**
compiling v1.12.0 or v1.12.3 and uploading.
run ./bin/px4 -s px4.config
 

**Expected behavior**
bmp280 #0 on I2C bus 2
mpu9250 #0 on I2C bus 2

**Log Files and Screenshots**
debian@beaglebone:~/px4$ sudo ./bin/px4 -s px4.config
INFO  [px4] mlockall() enabled. PX4's virtual address space is locked into RAM.
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
INFO  [parameters] BSON document size 127 bytes, decoded 127 bytes
  BAT1_V_CHANNEL: curr: -1 -> new: 5
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
bmp280 #0 on I2C bus 2 address 0x76
WARN  [SPI_I2C] UnknownApp: no instance started (no device on bus?)
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [drivers_board] Initializing librobotcontrol ...
INFO  [logger] logger started (mode=all)
INFO  [mavlink] using network interface SoftAp0, IP: 192.168.8.1
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 192.168.8.255
INFO  [px4] Startup script returned successfully

**Drone (please complete the following information):**
- Beaglebone Blue

**Additional context**
Add any other context about the problem here.

Ncerzzk, 2022-04-20:
can you show the `px4.config`?

for now, the default interface of mpu9250 is spi, so if you want to use i2c, using mpu9250_i2c

navunlimited, 2022-04-20:
> how about check the address of mpu9250 by `i2cdetect -y -a N` and using `mpu9250_i2c -I -a 0xXX start` ?

None of them work:
debian@beaglebone:~/px4$ sudo i2cdetect -y -a 2
Error: Can't use SMBus Quick Write command on this bus

However, I know the bus address 0x68 from using v1.11.3 version by printing out info when mpu9250 was called:
>>I2CSPIDriverBase::module_start is called
>>I2c bus is internal
>>I2C initialize opening I2C bus/device: /dev/i2c-2: #fd: 8
>>device_ID: 2385937 on I2C bus 2 at 0x68 by sq
>>mpu9250 initial sample rate: 1000
>>mpu9250 after change sample rate: 50
>>IMU9250 probe WHOAMI: 0x71

so try the second command:
pxh> mpu9250_i2c -I -a 0x68 start
ERROR [SPI_I2C] Bug: driver mpu9250_i2c does not pass the I2C address to I2CSPIDriverBase

I think the point is that MPU9250 from v1.11.3 was working. Then got broken after being removed or combined into invensens/mpu9250.
Not sure how many differences between the invensens/mpu9250 and the previous mpu9250 driver which is using librobotcontrol in beaglebone blue.

dagar, 2022-04-20:
@navunlimited I can take a closer look if anyone is actually using this board.

Can you try running `i2cdetect` within PX4?

dagar, 2022-04-20:
Can you try this on current master as well?

navunlimited, 2022-04-21:
@dagar Thank you for your time and effort.
I have tried compiling master head (after installing two more tools: kconfiglib and jsonschema) and uploaded to bb blue board. Then started px4 and tried i2cdetect. But no luck. It looks like i2cdetect can only run in linux, not in px4:

============================================
debian@beaglebone:~/px4$ sudo ./bin/px4 -s px4.config
[sudo] password for debian:
INFO  [px4] mlockall() enabled. PX4's virtual address space is locked into RAM.
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
INFO  [param] selected parameter default file eeprom/parameters
ERROR [param] open 'eeprom/parameters' failed (2)
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  SYS_AUTOSTART: curr: 0 -> new: 4011
  MAV_TYPE: curr: 0 -> new: 2
  BAT1_V_CHANNEL: curr: -1 -> new: 5
  BAT1_V_DIV: curr: -1.0000 -> new: 11.0000
INFO  [dataman] data manager file './dataman' size is 7866640 bytes
bmp280 #0 on I2C bus 2 address 0x76
ERROR [SPI_I2C] UnknownApp: no instance started (no device on bus?)
ERROR [gps] failed to instantiate object
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [mavlink] using network interface SoftAp0, IP: 192.168.8.1
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 192.168.8.255
INFO  [drivers_board] Initializing librobotcontrol ...
INFO  [logger] logger started (mode=all)
INFO  [px4] Startup script returned successfully
pxh> i2cdetect
Invalid command: i2cdetect
===============================================

tried  i2cdetect -l in debian:
========================================
debian@beaglebone:~$ i2cdetect -l
i2c-1   i2c             OMAP I2C adapter                        I2C adapter
i2c-2   i2c             OMAP I2C adapter                        I2C adapter
i2c-0   i2c             OMAP I2C adapter                        I2C adapter
===========================================

Based on v1.11.3 experience, I knew both bmp280 and mpu9250 are on i2c-2:
===========================================
debian@beaglebone:~$ i2cdetect -y -r 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- 76 --
===============================================

we can see bmp280 is @0x76, mpu is @0x68, which both match my previous post info from using v1.11.3.
Hope these info helps.

navunlimited, 2022-07-06:
@dagar any progress or need any help?

navunlimited, 2022-08-06:
@dagar I tried again on using SPI bus for adis16477 which was working on v1.11.3. But no luck on v1.12.0.
So, I guess i2c and spi drivers both have issues after updating to v1.12.0

###################################################
debian@beaglebone:~/px4$ sudo ./bin/px4 -s px4.config
INFO  [px4] mlockall() enabled. PX4's virtual address space is locked into RAM.
INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh px4.config 0
INFO  [parameters] BSON document size 387 bytes, decoded 387 bytes
  BAT1_V_CHANNEL: curr: -1 -> new: 5
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
bmp280 #0 on I2C bus 2 address 0x76
WARN  [SPI_I2C] UnknownApp: no instance started (no device on bus?)  <------ first one to start mpu9250
WARN  [SPI_I2C] UnknownApp: no instance started (no device on bus?)  <------- second one to start adis16477
INFO  [commander] LED: open /dev/led0 failed (22)
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on /dev/ttyS5 @ 460800B
INFO  [drivers_board] Initializing librobotcontrol ...
INFO  [logger] logger started (mode=all)
INFO  [logger] Logging FIFO data: increasing task prio and logging rate  
INFO  [logger] Start file log (type: full)
INFO  [logger] [logger] ./log/2022-08-06/09_10_58.ulg
INFO  [logger] Opened full log file: ./log/2022-08-06/09_10_58.ulg
INFO  [px4] Startup script returned successfully

################################################################

Another issue I found is that there is no IMU FIFO logging.
I was using px4_v1.11.3 stable version and I did not see any IMU FIFO message.
Any idea to make it work?
#################################
INFO [logger] logger started (mode=file)
INFO [logger] Logging FIFO data: increasing task prio and logging rate  <----- look like SDLOG_PROFILE = 771 setting is correct
INFO [logger] Start file log (type: full)
INFO [logger] [logger] ./log/2022-08-05/05_22_20.ulg
INFO [logger] Opened full log file: ./log/2022-08-05/05_22_20.ulg
###################################

no IMU FIFO files found by either using “ulog2csv” log analysis tool
[image]

or matlab’s ulogreader() then readTopicMsgs():
[image]

dagar, 2022-08-06:
@navunlimited ping me on slack to discuss?

I don't know what the problem is, but I'd be happy to get this fixed in the `main` branch, and then backported for the next stable point release (v1.13.1).

navunlimited, 2022-08-07:
@dagar , I just created my Slack account (squarechan with workspace name navunlimited) and invited you by using your email: daniel@agar.ca. My current location is Taiwan , so I guess we have 12 hours time difference if you are in Florida.
Looking forward.