PX4/PX4-Autopilot #19513: Relaxing the requirement for barometer presence/health in EKF initialization

spiderkeys, 2022-04-20:
**The issue:**
On one of our UUV vehicle frames, we have run into a situation where the EKF can not initialize because of the lack of a barometer which provides an initial Z position/velocity estimate. On this frame, the sole source of this information (besides the poor estimates from IMU), is an external INS system that does have Z information, but has all of its data plumbed in through the external vision interface, and which does not become fully functional/calibrated/healthy until 2-5 minutes after startup.

Within the current code-base, the lack of barometer causes the EKF to never initialize, regardless of whether the EV input is available, because Ekf::initialiseFilter() currently performs a hard `return false;` if there aren't an adequate amount of baro samples being delivered:
```c++
	if (_baro_counter < _obs_buffer_length) {
	 	// not enough baro samples accumulated
	 	return false;
	}

	// we use baro height initially and switch to GPS/range/EV finder later when it passes checks.
	setControlBaroHeight();
```
Because the EKF never initializes, other estimates which do have healthy sources available are never published (such as attitude). For our use-case, the vehicle itself is neutrally buoyant and naturally stable in Z, such that it is able to be flown in manual mode or stabilize mode without Z pos/velocity control. 

In addition to certain types of UUVs, this is probably also relevant for certain ground vehicles, where Z pos estimates may not be available/important depending on the FMU board being used and the use-case.

**Desired behavior:**
Ideally, the EKF should be able to initialize and start fusion for whichever estimate types it has valid inputs for. In this case, attitude and heading fusion should be able to fire up, while height fusion should be able to be deferred until an adequate healthy input is discovered. There are probably other scenarios besides the current baro behavior that this could apply to as well (i.e. allowing init without an attitude estimate, for vehicles/systems that don't care and may travel along a fixed axis somewhat like a 3D printer head).

The state of whether or not there is a healthy and active fusion process running should then be used to inform the validity of position estimates, such as `z_valid` and `v_z_valid` in `vehicle_local_position`, etc.

**The quick workaround**
I was able to achieve the basic desired behavior using the following quick and dirty approach:

1. Remove the barometer check in the EKF initialization process:
```c++
	// Allow initialization to succeed, even if there is no height estimate
	// The appropriate fusion method will be enabled dynamically
	// if (_baro_counter < _obs_buffer_length) {
	// 	// not enough baro samples accumulated
	// 	return false;
	// }

	// // we use baro height initially and switch to GPS/range/EV finder later when it passes checks.
	// setControlBaroHeight();
```
2. Change lpos.z_valid and lpos.v_z_valid to be determined by the control status flag of the EKF:
```c++
	// lpos.z_valid = !_preflt_checker.hasVertFailed();
	// lpos.v_z_valid = !_preflt_checker.hasVertFailed();

	// Z estimates are valid if we have a baro or external system that is actively running
	lpos.z_valid = (_ekf.control_status_flags().baro_hgt || _ekf.control_status_flags().ev_hgt );
	lpos.v_z_valid = lpos.z_valid;
```
3. Added a `stopBaroHgtFusion()` method which can be triggered by the loss of baro as a height estimate:
```c++
void Ekf::stopBaroHgtFusion()
{
	_control_status.flags.baro_hgt = false;
}
```
4. Updated Ekf::controlHeightFusion to dynamically enable/disable baro as a height fusion input. Ultimately `_baro_hgt_intermittent` is the value that ends up driving this dynamic decision:
```c++
void Ekf::controlHeightFusion()
{
	checkRangeAidSuitability();
	const bool do_range_aid = (_params.range_aid == 1) && isRangeAidSuitable();

	switch (_params.vdist_sensor_type) {
	default:
		ECL_ERR("Invalid hgt mode: %" PRIi32, _params.vdist_sensor_type);

	// FALLTHROUGH
	case VDIST_SENSOR_BARO:
		if (do_range_aid) {
			if (!_control_status.flags.rng_hgt && _range_sensor.isDataHealthy()) {
				startRngAidHgtFusion();
			}

		} else {
			if (!_control_status.flags.baro_hgt) {
				if (!_baro_hgt_faulty && !_baro_hgt_intermittent) {
					PX4_INFO( "Starting baro height fusion" );
					startBaroHgtFusion();

				} else if (!_control_status.flags.gps_hgt && !_gps_intermittent && _gps_checks_passed) {
					// Use GPS as a fallback
					startGpsHgtFusion();
				}
			}
			else
			{
				if (_baro_hgt_faulty || _baro_hgt_intermittent)
				{
					PX4_ERR( "Stopping baro height fusion" );
					stopBaroHgtFusion();
				}
			}
		}

		break;
```

I tested this by:
1. Starting the board with no baro driver running 
2. Checking that the EKF initializes and provides attitude information, and that the z_valid/v_z_valid flags in local position were false.
3. Starting the baro driver
4. Checking that the zpos estimate converged to the correct value and that z_valid/v_z_valid get set to true
5. Stopping the baro driver explicitly
6. Checking that the EKF continue to provide attitude estimates and that z validities get set to false again

There are probably other concerns/missing changes that need to be taken into consideration beyond this simple hack - open to ideas/feedback on the idea and the best approach to support these use-cases.

@dagar

bresch, 2022-04-21:
We would need to fuse some synthetic measurements to prevent an unbounded growth of the vertical channel covariance. To do so, we can fuse a constant fake altitude with a moderate observation variance, like we do when no horizontal aiding is available (see https://github.com/PX4/PX4-Autopilot/blob/master/src/modules/ekf2/EKF/fake_pos_control.cpp#L41). This would also help to stabilize the accelerometer bias estimate to prevent an uncontrolled divergence of the filter.

This is not only applicable to UUVs or UGVs but any drone where a barometer cannot be used or an autopilot that doesn't have this sensor.

bresch, 2022-04-21:
> 2. Change lpos.z_valid and lpos.v_z_valid to be determined by the control status flag of the EKF

This should be in addition to the `_preflt_checker.hasVertFailed()` because having the aiding source active doesn't mean that the fusion process is working properly

spiderkeys, 2022-04-21:
Thanks for the feedback.

> This should be in addition to the `_preflt_checker.hasVertFailed()` because having the aiding source active doesn't mean that the fusion process is working properly

I agree, though there seem to be some other intertwined dependencies that have to get resolved to make this possible for our frame. In our case:
- hasVertFailed() always returns the same result, because the variables it derives from are never updated
- those variables are never updated because PreFlightChecker::update() is never called
- PreFlightChecker::update() is never called because our vehicle is always considered to be "in air":
```c++
// calculate noise filtered velocity innovations which are used for pre-flight checking
	if (!_ekf.control_status_flags().in_air) {
		// TODO: move to run before publications
		_preflt_checker.setUsingGpsAiding(_ekf.control_status_flags().gps);
		_preflt_checker.setUsingFlowAiding(_ekf.control_status_flags().opt_flow);
		_preflt_checker.setUsingEvPosAiding(_ekf.control_status_flags().ev_pos);
		_preflt_checker.setUsingEvVelAiding(_ekf.control_status_flags().ev_vel);

		_preflt_checker.update(_ekf.get_imu_sample_delayed().delta_ang_dt, innovations);

	} else if (_ekf.control_status_flags().in_air != _ekf.control_status_prev_flags().in_air) {
		// reset preflight checks if transitioning back to landed
		_preflt_checker.reset();
	}
```
- We are always considered to be in air, because we don't run land_detector at all (doesn't make any sense for a UUV and was ultimately causing undesired behaviors, so we stripped it out and also pared down commander greatly).
```c++
const bool in_air = !vehicle_land_detected.landed;
```
All of that said, I agree that it *should* be checking for a healthy source, not just an active one. Any advice on how to reconcile the fact that some platforms may not use land_detector, and thus can't expect preFlightChecker to get updates in the current implementation?

On a related note, when I first started investigating this issue, it was somewhat surprising to me that there is a concept of pre-flight checks exist in the EKF. It seems like these types of checks and driving actions based off of them are more in the purview of `commander` and the flight mode state machine. I wouldn't expect EKF to be relying on pre-flight check results as the source of health/validity for input sources in functions like PublishLocalPosition(), since pre-flight checks currently only happen pre-flight, and aren't continuously evaluated.

bresch, 2022-04-22:
> Any advice on how to reconcile the fact that some platforms may not use land_detector, or more generally are just never considered to be in a landed state, and thus can't expect preFlightChecker to get updates in the current implementation?

Can't you base the `landed` on arming (i.e.: `landed = !armed`)

dagar, 2022-04-22:
>  Any advice on how to reconcile the fact that some platforms may not use land_detector

This doesn't help you with the preflight checks, but at a minimum I've been trying to ensure things still basically function without a land detector entirely. https://github.com/PX4/PX4-Autopilot/commit/11f617ca9be36684388c2d1c2a2e1f6698597e44

bresch, 2022-10-26:
fixed here: https://github.com/PX4/PX4-Autopilot/pull/20021/files

closed 2022-10-26 by bresch