PX4/PX4-Autopilot #19538: ODOMETRY giving opposite X and Y data in qgc

bhaskar-glitch, 2022-04-26:
Hi,
I'm using Pixhawk 2.4.8 with 1.12.3 firmware.Im using VIO and the data in mavros/vision_pose/pose and all the mavros position topics are fused properly all the x and y coordinates are correct.But when I'm visualising the data in MAVLINK INSPECTOR of ODOMETRY the x and y position is opposite to what being I'm unable to find what's the actual issue.
Please help

Jaeyoung-Lim, 2022-04-26:
@bhaskar-glitch I dont think this is an issue

mavros(rostopics) and mavlink normally use different frames. ROS uses ENU(East North Up) and mavlink ises NED(North East Down). The conversions are done inside mavros

Kihoon-Shin, 2023-04-05:
Hello, @bhaskar-glitch I have a same problem..
Did you solve it?

bhaskar-glitch, 2023-04-05:
> Hello, @bhaskar-glitch I have a same problem.. Did you solve it?

I guess you have been using Camera Imu doing the calibration. Just change the camera imu to pixhawk imu topic which /mavros/imu/data and calibrate camera's properly. Because in IMU of camer Y is forward and in pixhawk IMU X is forward.

Kihoon-Shin, 2023-04-06:
@Jaeyoung-Lim Thank you for your reply.
I'm following VIO tutorial in PX4 document and It says "Moving the vehicle in forward direction, should increase the position's x coordinate. While moving the vehicle to the right should increase the y coordinate." in MAVLINK inspector.
but when I move a drone back and forth, the y value is changing.
and when I move a drone side to side, the x value is chnaing.
https://user-images.githubusercontent.com/82350419/230252796-95dd2b1b-f0d2-46d1-8446-38074a2c8365.mp4

Jaeyoung-Lim, 2023-04-06:
@Kihoon-Shin are gou sure you have configured the autopilot orientation correctly? Is this without sending vio?

Kihoon-Shin, 2023-04-07:
@Jaeyoung-Lim Ok it's solved! After arranging the orientation to a unit quaternion (1,0,0,0) correctly, The problem was solved. Thank you.

closed 2025-09-27 by Jaeyoung-Lim