PX4/PX4-Autopilot #19541: RC not responding when vehicle is powered up before the RC

AhmedElsafy, 2022-04-26:
I have a FrSky RC and receiver, vehicle is running 1.13-alpha, when vehicle is powered up then the RC, I can see on QGC RC channel is being detected but no response from the vehicle. It can be safe in most cases as arming the vehicle will not happen, but I have sent the vehicle to Mission mode and armed it from QGC thinking that the RC is operating then I couldnt take over control to bring back the vehicle.

taileron, 2022-04-27:
the behaviour of the parameter COM_RC_IN_MODE has slightly changed there are now more possibilities regarding priority and fallback there are now more options regarding priority and fallback. I just have to use the value 3, unfortunately I can no longer find that in the documentation.

taileron, 2022-04-27:
QGC daily shows all information behind the values and 3 is the new default setting.