PX4/PX4-Autopilot #19566: when i do : make px4_sitl_default gazebo , it put

tangqi54188, 2022-05-01:
-- PX4 version: v1.11.0-beta1
-- PX4 config file: /home/tangqi/uav_project/volans/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.17") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-107-g1813034
-- Configuring done
-- Generating done
CMake Warning (dev):
  Policy CMP0058 is not set: Ninja requires custom command byproducts to be
  explicit.  Run "cmake --help-policy CMP0058" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  This project specifies custom command DEPENDS on files in the build tree
  that are not specified as the OUTPUT or BYPRODUCTS of any
  add_custom_command or add_custom_target:

   test_mixer_multirotor

  For compatibility with versions of CMake that did not have the BYPRODUCTS
  option, CMake is generating phony rules for such files to convince 'ninja'
  to build.

  Project authors should add the missing BYPRODUCTS or OUTPUT options to the
  custom commands that produce these files.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Build files have been written to: /home/tangqi/uav_project/volans/Firmware/build/px4_sitl_default
[3/733] git submodule src/drivers/gps/devices
[22/733] git submodule src/lib/ecl
[23/733] git submodule mavlink/include/mavlink/v2.0
[28/733] git submodule Tools/sitl_gazebo
[34/733] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
[... 103 lines cut ...]
-- Found GStreamer: GSTREAMER_INCLUDE_DIRS;GSTREAMER_LIBRARIES;GSTREAMER_VERSION;GSTREAMER_BASE_INCLUDE_DIRS;GSTREAMER_BASE_LIBRARIES (Required is at least version "1.0") 
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- Found MAVLink: /home/tangqi/uav_project/volans/Firmware/mavlink/include (found version "2.0") 
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 3.0.0
-- Gazebo version: 9.19
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tangqi/uav_project/volans/Firmware/build/px4_sitl_default/build_gazebo
[729/733] Performing build step for 'sitl_gazebo'
ninja: error: 'cmake_object_order_depends_target_OpticalFlow_', needed by 'cmake_object_order_depends_target_gazebo_opticalflow_plugin_', missing and no known rule to make it
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/tangqi/uav_project/volans/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:200: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

help me please

julianoes, 2022-05-01:
Can you do `git submodule update --init --recursive` and try again?

tangqi54188, 2022-05-02:
i try it ,
and make px4_sitl_default gazebo again 
that put out :
-- PX4 version: v1.11.0-beta1
-- PX4 config file: /home/tangqi/uav_project/volans/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-255-g47624a0
CMake Error at CMakeLists.txt:402 (add_subdirectory):
  add_subdirectory given source "src/modules/px4_simulink_app" which is not
  an existing directory.

-- Configuring incomplete, errors occurred!
See also "/home/tangqi/uav_project/volans/Firmware/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /home/tangqi/uav_project/volans/Firmware/build/px4_sitl_default is not a directory
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

julianoes, 2022-05-15:
Have you resolved this in the meantime? Is this recent master or something older?