PX4/PX4-Autopilot #19567: Pool Timeout - PX4 Simulator Gazebo

JC-hub, 2022-05-02:
**Describe the bug**
I am trying to control each rotor of a quadrotor directly, so I write a `overwrite.main.mix`, which is similar to the pass.aux.mix, and apply it to both a real quad airframe `4001_quad_x` and a simulated quad `if750a`. This mixer works well in the real quad but occurs `ERROR [simulator] poll timeout 0, 22` in the simulator. 

**To Reproduce**
I commit my code on GitHub, which was developed on branch `release/v1.12`:
https://github.com/ErcBunny/PX4-Autopilot/tree/overwrite-main-v1.12-issue
1. To reproduce the error:
    - run the simulation: `make px4_sitl gazebo_if750a`
    - pub actuator_controls_1 to overwrite mixer: `px4_simple_app 1 -1 -0.5 0.5 1 `
2. Apply it on a real quadrotor: 
    - ` make cubepilot_cubeorange `
    - `px4_simple_app 1 -1 -0.5 0.5 1 `

**Log Files and Screenshots**
- Screenshot about the error occurring in simulation.
[image]

- Screenshot about the mixer working well in a real quad. 
[image]

**Additional context**
I find a similar issue about 'ERROR [simulator] poll timeout 0, 22' here: 
https://github.com/PX4/PX4-Autopilot/issues/19203 
and a disscussion: 
https://discuss.px4.io/t/px4-poll-timeout/20484 

Thanks!

zhikh1, 2023-01-13:
Did you manage to solve the problem? I am facing the same behavior with my custom model

julianoes, 2023-01-13:
If you are publishing actuator_controls directly, you need to use SITL without lockstep:

https://docs.px4.io/main/en/simulation/#disable-lockstep-simulation

schefferac2020, 2024-05-29:
My issue was not including the correct imu, barometer, and magnetometer models

julianoes, 2024-05-29:
@ZhaiMen-Hub can this be closed? Did you find a fix?

ZhaiMen-Hub, 2024-05-30:
No. Because we didn't find a solution at that time, we this attempt is terminated in June 2022

julianoes, 2024-05-30:
Ok thanks for the note.

closed 2024-05-30 by julianoes