PX4/PX4-Autopilot #19572: Help

cjychenjiayi, 2022-05-03:
I've install all the document under the guideline， when I type make px4_fmu-v4_rtps, it shows  as below:
Makefile:527: *** "Make target px4_fmu-v4_rtps not found. It either does not exist or px4_fmu-v4_rtps cannot be the first argument. Use 'make help|list_config_targets' to get a list of all possible [configuration] targets.".  Stop.
make px4_fmu-v4 , make px4_sitl_rtps can both run successfully, I don't know why it happened
my system is Ubuntu 20.04
Looking forward to help, thx

chenrc98, 2022-05-03:
Create a file in `PX4_Source/boards/px4/fmu-v4` named `rtps.px4board` and put the following line`CONFIG_MODULES_MICRORTPS_BRIDGE=y` into the file. Then try `make px4_fmu-v4_rtps`

cjychenjiayi, 2022-05-04:
Thx for your help, it worked!

PR #20203 (2022-09-15): add v4 rtps build
## Describe problem solved by this pull request
Fixes #19572. We should have an `rtps` target at least for `px4_fmu-v4` because the [official documentation](https://docs.px4.io/main/en/middleware/micrortps.html) uses `make px4_fmu-v4_rtps` as an example.

## Describe your solution
Add a `rtps.px4board` to `board/px4/fmu-v4`.
closed 2022-05-13 by cjychenjiayi