PX4/PX4-Autopilot #19574: fast blinking purple lights, commands do not respond, mavlink console does not work, high cpu?

dlech, 2022-05-03:
I'm trying to get a Cube Black to work with a RPi as an onboard computer. I got this to work once, but after I rebooted the autopilot, the status lights will start blinking fast purple after short while (usually the light would turn from blue flashing (ready, no GPS) to green flashing (ready, GPS) at this time but it goes from green to purple instead). [The manual](https://docs.px4.io/master/en/getting_started/led_meanings.html) does not have this listed. Also there is a flast-blinking orange (or red? hard to tell since there is also slow blinking green light) on the autopilot itself. I haven't found any documentation on the meaning of the lights on the autopilot itself.

Also, when this happens, the autopilot no longer responds to most commands. QGroundControl will show errors that XYZ_COMMAND failed or "failed to reboot" or something like that depending on which command. Furthermore, when this happens, the Mavlink console no longer works.

I tried this with firmware versions 1.12.3, 1.13 beta 1 and master all with the same results.

The first way I figured out how to make the error go away is to flash the Ardupilot firmware using Mission Planner then flash the PX4 firmware again using QGroundControl. I'm guessing that flashing with Mission Planner erases more than QGroundControl does? 
Unfortunately, the "fix" doesn't last very long. I'm able to use things normally once, but then after an additional reboot, the problems come back again.

Then I tried downgrading to firmware v1.11.3 and the problem goes away. So this appears to be firmware issue.

I'm not sure how to provide any debugging information or logs since nothing works when the error happens. I did once see a high CPU error pop up in QGroundControl when this is happening, so maybe that is a hint (getting stuck in an infinite loop in a CPU bound task?).

dagar, 2022-05-03:
Which airframe did you select (SYS_AUTOSTART)? Did you configure anything else like extra mavlink instances?

dlech, 2022-05-03:
`SYS_AUTOSTART` is 4001 (generic quadcopter).

In addition to the required setup (airframe, sensors, radio, flight mode, power) which was already working with several flights made, the new settings are: `MAV_1_CONFIG1` is set to  `TELEM 2`, additional `MAV_1_*` parameters are left at default (`_MODE` = `Onboard`, `_RATE` = `0`, tried both `_FORWARD` = `0`). `SER_TEL2_BAUD` is set to 57600 8N1.

Owen-K, 2022-05-27:
I am seeing this same issue on multiple cube blacks as well. It seems as if it might be a memory issue but I am not completely sure of this.

mcsauder, 2022-05-27:
@dlech , I had this happen the other day.  I fixed my issue by removing the SD card, erasing all contents (via another computer), replacing it into my autopilot, and then re-flashing the autopilot. Let us know if this fixes your problem.

dagar, 2022-05-27:
> @dlech , I had this happen the other day. I fixed my issue by removing the SD card, erasing all contents (via another computer), replacing it into my autopilot, and then re-flashing the autopilot. Let us know if this fixes your problem.

@mcsauder FYI you can trigger a format creating a `.format` file on the SD card and rebooting.

``` Console
nsh> echo "" > /fs/microsd/.format
nsh> reboot
```

[image]

dlech, 2022-05-27:
I actually tried this already (I used a new SD card instead of erasing the old one), but it didn't seems to make a difference in my case.

dagar, 2022-05-27:
> I actually tried this already (I used a new SD card instead of erasing the old one), but it didn't seems to make a difference in my case.

Sorry, that was more of a response to Mark. Reading your original comment it sounds like high cpu usage.

Do you have an airframe selected? Can you get to a console and share the output of running `top`?

I quick way to reduce cpu usage is reduce the inner loop rate (eg `param set IMU_GYRO_RATEMAX 400`), disable unused mavlink instances, etc, etc. I can provide more specific advice after we figure out exactly what's happening.

mcsauder, 2022-05-27:
Thanks for the advice @dagar!

@dlech , The way I figured out I had an SD card issue was by having a connection to the nsh console during boot, then watching how far the boot sequence proceeded.  Can you post the boot sequence output here?

dagar, 2022-05-27:
@dlech can you give this a try? https://github.com/PX4/PX4-Autopilot/pull/19730

dagar, 2022-05-27:
The mavlink USB instance is a huge resource pig, simply skipping the USB connection if you don't need it (with master or v1.13) will save a huge amount of memory.

dlech, 2022-05-27:
> can you give this a try? #19730

I just tried building and installing the `px4_fmu-v3` firmware from c27b9f3b1e.

After rebooting, the problem persisted, but the USB was still plugged in from updating the firmware. So the changes still don't free up enough space to have USB + telemetry MAVLINK + onboard computer MAVLINK (serial/UART) .

I unplugged USB and rebooted again and I am actually able to get a shell. So here is `top` without USB connected:

```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE FD
   0 Idle Task                  247235 38.590   224/  512   0 (  0)  READY  3
   1 hpwork                          0  0.000   332/ 1264 249 (249)  w:sig  3
   2 lpwork                          0  0.001   588/ 1616  50 ( 50)  w:sig  3
   3 init                            0  0.000  2332/ 2928 100 (100)  w:sem  3
   4 wq:manager                      0  0.000   556/ 1256 255 (255)  w:sem  3
 908 log_writer_file                 0  0.000   364/ 1176  60 ( 60)  w:sem  3
  62 wq:hp_default                  19  1.214  1080/ 1904 237 (237)  w:sem  3
  77 dataman                         0  0.007   820/ 1208  90 ( 90)  w:sem  4
  85 wq:lp_default                  18  1.122  1068/ 1920 205 (205)  w:sem  3
 297 wq:uavcan                      56  3.445  3092/ 3624 236 (236)  w:sem  3
 331 wq:SPI1                        69  4.303  1652/ 2336 253 (253)  w:sem  3
 344 wq:SPI4                       148  9.174  1676/ 2336 250 (250)  w:sem  3
 550 wq:nav_and_controllers        100  6.177  1308/ 2240 242 (242)  w:sem  3
 551 wq:rate_ctrl                  140  8.679  1468/ 3152 255 (255)  w:sem  3
 554 wq:INS0                       146  9.032  4244/ 6000 241 (241)  w:sem  3
 559 commander                      35  2.123  1236/ 3224 140 (140)  w:sig  5
 752 gps                             2  0.124  1188/ 1680 205 (205)  w:sem  4
 796 mavlink_if0                    51  3.214  1820/ 2728 100 (100)  READY  6
 797 mavlink_rcv_if0                 9  0.584  1732/ 4712 175 (175)  w:sem  6
 822 mavlink_if1                   102  6.273  1892/ 2728 100 (100)  w:sig  4
 823 mavlink_rcv_if1                 8  0.538  1732/ 4712 175 (175)  w:sem  4
 850 navigator                       3  0.227  1196/ 1920 105 (105)  w:sem  6
 901 logger                          8  0.510  2548/ 3648 230 (230)  w:sem  3
 916 mavlink_shell                   0  0.000  1084/ 2024 100 (100)  w:sem  3
 923 top                            24  1.738  2988/ 4080 237 (237)  RUN    3

Processes: 25 total, 3 running, 22 sleeping
CPU usage: 58.49% tasks, 2.92% sched, 38.59% idle
DMA Memory: 5120 total, 1024 used 1536 peak
Uptime: 554.876s total, 247.235s idle
```

and `free`:

```
                   total       used       free    largest  nused  nfree
        Umem:     231600     218080      13520      13376   1023      9
        Prog:    2097152    1966080     131072          0     24      1
```

dagar, 2022-05-27:
Thanks @dlech, I just noticed you have uavcan running, are you actually using it? It's probably even more of a pig than the Mavlink USB instance. 

To be honest one of the reasons memory has gotten pretty tight on these older boards is that most of us have moved onto newer STM32F7 or STM32H7 that have 2x-4x the available RAM as the pixhawk 2. The pixhawk 4 has been out for more than 4 years I believe. I'd be happy to help work through a few optimizations to keep these usable, but it's definitely a losing battle at this point.

dlech, 2022-05-27:
Yes, using it for a Here 2 GPS - although I suppose it is technically possible to change it to use UART instead.

Upgrading the autopilot is an option too, so probably more reasonable than spending time on trying to keep trimming things down.

Thanks for the feedback, it has been very helpful in understanding the issue.

dlech, 2022-05-27:
I'm going to go ahead and close the issue since it is now well understood and there isn't a specific "fix":

In summary:

The problem is that Cube Black (and other older hardware) don't have enough RAM for firmware v12 to use USB + UAVCAN + 2 more MAVLINK instances. Workaround is to downgrade firmware or use less resources by not using USB, disabling UAVCAN, etc. or to upgrade to a newer autopilot with more memory.

closed 2022-05-27 by dlech

dagar, 2022-05-27:
> Yes, using it for a Here 2 GPS - although I suppose it is technically possible to change it to use UART instead.

Switching that back to serial would probably be the easiest thing to do short term if you have no immediate benefits from CAN (wiring, etc).

On the other hand you could also disable the onboard gps driver to save a bit of memory. Check the `GPS_1_CONFIG` and `GPS_2_CONFIG` parameters or simply `gps stop` if you want to immediately check the memory difference with `free` before and after.