PX4/PX4-Autopilot #19577: Dynamic Control allocation(Master branch): flight issues

dookei, 2022-05-04:
Hi there,
I have recently moved to v1.13 so that I can use the control allocator. The objective was to have some flexibility on the design and control from several pwm actuators(servos, lights, switchs etc). The use was described on [issue 19190](https://github.com/PX4/PX4-Autopilot/issues/19190). This works just as expected.

Now, it comes the part that I need to test the flight stability. The UAV can't simply fly stable enough to even get it one meter over the ground. I tried several things...PID tunning(this was already done before)...recheck for vibration issues... etc 
Vibrations look horrible by the way. Which is weird, since much hasn't changed since I moved to v1.13(master). 
[image]
[image]

**Back to v1.12.3**
In order to exclude any hardware issues, I uploaded the v1.12.3 (that was working fine before). This still works fine. There are no significant vibrations. The attitude control also looks smoother(still needs some fine tuning but flyable and responsive)
[image]
[image]

The airframe: 
This is a dodecarotor, but on the FC perspective it seen as a quadcopter. I also added an external compass because there was some issues with the internal/FC one. 
[image]
https://tuenkers-nickel.de/wp-content/uploads/2022/01/TNK-Transportdrohne_mag_001.jpg

I would like to keep working on the v1.13, since this allows me to control the rest of the non-flight relevant actuators.
Is this even an issue related to Dynamic Control Allocation or is it related to the Attitude & Rate Controller?
Any thoughts on what might have changed since v.1.12 that might have influence on the flight stability?
I would appreciate if you could shed some light on this issue.

dagar, 2022-05-05:
> the UAV can't simply fly stable enough to even get it one meter over the ground

Do you have this log?

> This is a dodecarotor, but on the FC perspective it seen as a quadcopter. I also added an external compass because there was some issues with the internal/FC one.

Do you have to connect each motor to the flight controller directly?

dagar, 2022-05-05:
Can you try setting a notch filter at 120 Hz (IMU_GYRO_NF_FREQ 120) and see if you can get a bit more of an actual flight log? 

[image]

dookei, 2022-05-05:
> Do you have this log?

on this log you can see that. It got almost a meter and then crashed. Luckly I have a safety net ;-) [https://logs.px4.io/plot_app?log=1ab69420-9f27-4a27-8fe4-45cf037ca8c3](https://logs.px4.io/plot_app?log=1ab69420-9f27-4a27-8fe4-45cf037ca8c3)
> Do you have to connect each motor to the flight controller directly?

Each motor is controlled over DSHOT.There are 4 sets each with 3 motors and ESC. But the signal to every cluster(of three) as the same signal. Therefor for the FC is seen as 4 motors only.

> Can you try setting a notch filter at 120 Hz (IMU_GYRO_NF_FREQ 120) and see if you can get a bit more of an actual flight log?

I can give it a try tomorow. But this high angular accelaration, at 120 HZ, only happens in the ground when the engines are spinning at low rpm(armed no throttle). You can see that in the first log( where it got of the ground for a few secs) this doesnt seam to be an issue.
What I find weird is that on both PX4 versions( v.1.12.3 and v1.13.0) they both have the same parameters values. Why do they measure different vibrations when only armed?
V1.12.3
[image]

v1.13.0
[image]

dagar, 2022-05-05:
> I can give it a try tomorow. But this high angular accelaration, at 120 HZ, only happens in the ground when the engines are spinning at low rpm(armed no throttle). You can see that in the first log( where it got of the ground for a few secs) this doesnt seam to be an issue.

Don't bother with that notch, it was just a guess from the minimal ground log, it's definitely not the issue in the longer flight.

dagar, 2022-05-05:
FYI your rollrate and pitchrate gains did change a bit between those logs.

kP: 0.108 -> 0.075
kI: 0.0985 -> 0.056
kD: 0.0023 -> 0.0009

dookei, 2022-05-05:
> FYI your rollrate and pitchrate gains did change a bit between those logs.

Yes you are right. I changed them a littel after some fly and crashs... But I also tested with the same gains. 

>What's going on with your battery?
Dont mind that ;-). I am not using any battery. Its a tethered setup. One of the PSUs is not so stable... but that is not the issue.

dagar, 2022-05-05:
I believe the scaling of the vibration metrics changed and the thresholds need to be adjusted in flight review. It was previously based on delta angle and delta velocity, but changed to angular velocity and linear acceleration to keep it consistent at different integration rates and avoid false positives due to jitter.

For your new log it's hard to see what's happening because it only has about 4 seconds of flight. Your motors are saturating with the relatively large yaw error. I would temporarily decrease MC_YAWRATE_MAX drastically (200->40) and see if you can get a longer flight. Someone else might have better advice from this limited data.

dookei, 2022-05-06:
@dagar many thanks for you help. It really helps to have a second pair of eyes on this.
The the roll, pitch and yaw  where changed to 100,100 and 40 respectively. This doesn't have much influence. It looks to me  a little like "toilet bowl effect". What do you think?
Here is the log. I sent you the video per Wetransfer(hope its okay like that, was not working here).
https://logs.px4.io/plot_app?log=23180a36-911f-4492-946c-d85be97c37cf

I disabled the internal compass to make sure the external is being used.The internal is quite noisy.
https://github.com/PX4/PX4-Autopilot/blob/1f3a78b535001258a2b3c400c1bf7baab8702f30/boards/holybro/durandal-v1/init/rc.board_sensors#L27

I also tried reducing the K and PID rate for RPY. Still no significant change in there. Do you think the Dynamic control allocation might have any influence on it?

In parallel I am adding an extra internal magnetometer(the RM3100) on the SPI6. To be tested later on. I already soldered it ...still need to find time to test it. 
If anyone has interest this how it looks like. 
[image][image] [image][image]

closed 2022-05-16 by dookei