PX4/PX4-Autopilot #19611: h743-slim v1.5 config

matveylapin, 2022-05-06:
Matek h743-slim config update to the current version of the board. Changed the second imu to icm42605 , solved the problem with horizon calibration.
https://github.com/matveylapin/PX4-Autopilot/commit/182d2cbafb2df5ebddbd04df0a4ffd50ae8f1dc2

dagar, 2022-05-07:
Can you open a pull request? https://github.com/PX4/PX4-Autopilot/compare

matveylapin, 2022-05-07:
I was beaten by a couple of hours(((((
https://github.com/PX4/PX4-Autopilot/commit/9c12c2a1520186f565662acfc25f87da2925c5a5

dagar, 2022-05-09:
> I was beaten by a couple of hours((((( [9c12c2a](https://github.com/PX4/PX4-Autopilot/commit/9c12c2a1520186f565662acfc25f87da2925c5a5)

Great, let me know if you find anything else missing on this board.

Thanks

closed 2022-05-09 by dagar